supermerill/SuperSlicer #1971: 'infill before perimeters' setting is greyed out unexpectedly

blex-max, 2021-11-28:
**Describe the bug**
The 'infill before perimeters' setting is greyed out and not changeable despite the value of the setting affecting the print. Using the lock icon, the setting can be changed from on to off and the resulting gcode changes accordingly as can be seen in the preview. I would like to be able to change this setting at will and it seems like it shouldn't be greyed out given it's value affects the print.

**>> Project File <<**
[cube_infill_before_walls_bug.zip](https://github.com/supermerill/SuperSlicer/files/7613795/cube_infill_before_walls_bug.zip)

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Version: 2.3.57.5

Thanks for all the work you do!

neophyl, 2021-11-28:
Seems to be bound by having at least 1 bottom layer.  With 1 bottom layer set you can change it at will.  Which does seem a little strange as I can't see any reason why you couldn't print like that with no bottom layers.

supermerill, 2021-12-01:
It's currently activated if any solid fill is detected.
100% sparse fill isn't really a solid infill.

But i don't see why it's deactivated ether way, as the sparse infill is affected.

blex-max, 2021-12-04:
Thanks - are you suggesting that if I want a solid part I should not be using 100% sparse fill? If I instead turn on solid infill every layer I get weird layers of internal bridging infill etc. As always, thanks for all the work :)

supermerill, 2021-12-04:
There is many way to create a solid object. None are really wrong.

closed 2021-12-22 by supermerill