supermerill/SuperSlicer #1976: Support Blocker

MakerPay, 2021-11-29:
### Discussed in https://github.com/supermerill/SuperSlicer/discussions/1975

<div type='discussions-op-text'>

<sup>Originally posted by **MakerPay** November 28, 2021</sup>
To start I would like to say I love SuperSlicer and it has worked great for my needs till now.  Not sure if this is getting posted in the correct place so if not please direct me to the proper place.  I am working on a product and I think something is broke in Super Slicer.....or I just cannot get it to work.  I am trying to put two objects in the same area with one hanging over the other.  I am using the lower object in lou of using support.  My hope was to get the support to build on top of the lower object for the upper but it runs the support all the way through the lower object to the build plate.  I tried to use the support blocker slab but it has no effect.  I can however use the support blocker cube and it will block support but as soon as I change the shape of that cube its ability to block the support goes away.  What I want to do works in idea maker but I really do not want to learn how to use that program.  Help?  Thanks in advance
[image]
[image]

.</div>

neophyl, 2021-11-29:
What you are posting about is normal operation for Super Slicer and Prusa Slicer.
As each Object is sliced individually then when you overlap them you get 2x extrusions in the same space, bad things happen.  There are no checks for this, its the users responsibility.
The same is true for each Objects support too as its sliced with each object.

What you as a user need to do is to merge the Objects.  Once its a single Object with multiple Parts then it will slice correctly and also stop putting support through the lower model.  You can select both objects and right click>Merge.

MakerPay, 2021-11-29:
Man I am sorry to have wasted your time but super happy you just solved my problem.  That just knocked 1.5 hour off my print time and also the slice looks great.  Thank you very much.

closed 2021-11-29 by MakerPay