supermerill/SuperSlicer #1977: First Layer Infill Flow Ratio

Alexander-T-Moss, 2021-11-29:
**Is your feature request related to a problem? Please describe.**
I always find that the perimeters on the first layer go down nice and smooth but the infill will always look over extruded.

**Describe the solution you'd like**
I'd like a new setting to modify the Flow Ration/Extrusion Multiplier for the first layer's infill.

**Describe how it would work**
Add a new parameter in _Width & _Flow_/_Flow_ called "First Layer Infil" that would allow you to change the extrusion amount of just the infill, similar to "top fill"

**Describe alternatives you've considered**
I've tried increasing first layer and em but still get over extruded infill, and also played around with z offset but to get well extruded first layer infill, the perimeters don't stick very well, if not at all.

**Additional context**
:)

supermerill, 2021-12-01:
I don't see why or how. Did you have some good pictures that depict the problem?

Alexander-T-Moss, 2021-12-01:
I'm almost always finding my first layer perimeters will look perfect yet the infill will come out over extruded. I've tried adjusting the z offset but found that making the infill look fine leads to the perimeters showing signs of being too high (on both textured and smooth sheets) Below you'll see the perimeters being near perfect but the infill being over extruded.

[image]

supermerill, 2021-12-01:
Thanks.
If it's not a visual effect, your perimeter seems also very over-extruded, as it seems that there is a big ridge at the side:
[image]

If you have some difficulties to make the first layer stick, i advise to increase the width of the first layer first instead of over-extruding (because too near or too much flow).

Alexander-T-Moss, 2021-12-01:
That's just the reflection of the light as it's coming in from the side

Alexander-T-Moss, 2021-12-01:
I'll play around with the width of the first layer and see if that improves things

Alexander-T-Moss, 2021-12-01:
I'll bump the z offset too and see if it can go higher :)

ipatch, 2022-10-07:
this is something i've noticed as well. printing a rather large flat box. and i've dialed my flow back from 100% to 90% while the printer is printing the first layer infill.

how would adjusting the first layer width help with this? make it smaller or wider?

i'll attach some links to some examples of what i'm talking about.

https://user-images.githubusercontent.com/613805/194625862-6822cd1a-cd63-416b-bdb0-c365f36964dd.JPG

https://user-images.githubusercontent.com/613805/194625890-fcad8598-72a9-4ac4-89bb-2a0abe0cd8b1.JPG

https://gist.github.com/ipatch/9348354f85113b019e2be01c13759220?permalink_comment_id=4327836#gistcomment-4327836

---

and adjust my `z_offset` in klipper from `0.3` to `0.2` has seemed to help some.

ipatch, 2022-10-08:
tweaked a few settings. i think adjusting the first layer height from `0.20` to `0.25` helped some with issues above. this is what i'm getting now.

[image]

---

<video src="https://user-images.githubusercontent.com/613805/194723070-d6232203-0176-4f0f-9d5b-88e5afde70aa.MOV">

</video>

Alexander-T-Moss, 2022-10-18:
I'm considering making a post processing script as a proof of concept so will see how that goes and post results