supermerill/SuperSlicer #1978: Fan speed option for Support Interface(s)

gpread, 2021-11-29:
Supports under overhangs and bridges can be difficult to remove.

This is especially the case using some materials where low or no fan, higher ambient or nozzle temperatures are desirable.

Therefore I believe it could be helpful to cool the support interface layer(s) prior to the object layers above, so as to then limit their adhesion to the support structure. 

I imagine this might be relatively easy to implement just as another fan speed option along with "bridges", "top", "external perimeter", etc.

neophyl, 2021-11-29:
What contact z distance are you using ?  Also with what material ?  Contact z makes a massive difference in how difficult support is to remove and the defaults aren’t great.

supermerill, 2021-12-01:
duplicate of #937

gpread, 2021-12-01:
Typically I have used 0.2mm contact Z and been using PETG, but have experimented with other values for contact Z and not really found a setting where support works and its easily removed.