supermerill/SuperSlicer #1979: Filament Temperature modifier errors and blocks g-code generation with spiral vase mode

gpread, 2021-11-29:
2.3.57.6

Creality CR-10V3 (modified), Marlin 2.0.9.2
Windows 10 - 20H2 (19402.1165)

I have a print where spiral vase mode is enabled.  I can add a height modifier with a filament temperature setting. Which defaults to temperature of 0 (zero). 

As soon as I set it to a positive/reasonable value, Susi re-evaluates (maybe tries to re-slice?) and reports errors.

In the bottom right of preview pane it presents an error dialog indicating "The Spiral Vase option can only be use when printing single material objects" and the action button (Slice Now/Export G-Code) text changes to "invalid data".

Steps to reproduce the behavior:

1. Load the project file (attached below)
2. The error indication should be shown immediately (as seen in screenshot below)
3. Select the Filament modifier in the right side-bar
4. Alter the temperature value to 0 and then use the up/down selector buttons to toggle between 0 and 1

[image]

gpread, 2021-11-29:
[Cube-20-20-20_SpiralVase_SusiLayereheightTempMod_Issue#1979.zip](https://github.com/supermerill/SuperSlicer/files/7618070/Cube-20-20-20_SpiralVase_SusiLayereheightTempMod_Issue.1979.zip)

supermerill, 2021-12-01:
duplicate of #738

will modify taht next time i'll work on spiral thign. there is some issues on it.