supermerill/SuperSlicer #1981: Support below perimeters

Fabface, 2021-11-29:
**Is your feature request related to a problem? Please describe.**
When printing objects , which include some sort of pocket in the top surface (like the "Z" in the XYZ calibration cube), the perimeters of this pocket are not being supported before the first visible perimeter. It is standard in every other slicer I have tried to support the perimeters, or extend the support/infill area further and I don't see any point of having it not like that. I assume i must have overseen an option somewhere, though I've spent the last hour to find it, if so - please help me finding it.
[image]
[image]

**Describe the solution you'd like**
Add perimeters to the infill layers in pocket features (indicated in green)
[image]

**Describe alternatives you've considered**
Extend the infill area to support the perimeters above.

neophyl, 2021-11-30:
Normally supporting dense infill is applied. What version are you using ?  Also I assume that’s a normal calibration cube ? But it would be helpful to attach a saved project file so your settings can be examined.

Fabface, 2021-11-30:
Hi,

the version is 2.3.57.6. Please find the project file attached.

[xyzCalibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/7623976/xyzCalibration_cube.zip)

neophyl, 2021-11-30:
The setting No solid Infill Over:xx is causing issues but that's not all of it.  If you set it to 0 then it does an extra layer of bridge infill to support the area you outlined in your second pic which was what I was expecting.  That will help for most of it and that setting is quite new so I'm not surprised it has weird interactions/effects.

However even with that change the perimeters around the Z, while mostly supported on the infill still aren't on the corners which will droop and possibly show up on the top of the print as depressions.  

This is one for SuperMerill to have a look at I think as there could be further improvements made.

supermerill, 2021-12-01:
That's because the external_infill_margin is set to 0

Fabface, 2021-12-01:
That did fix it indeed, thank you. From the description of this parameter it appeared to me that this would extend the regular infill "into" the parameters - like overlap them, to better connect them or something like that, which I don't want, so I set it to 0.

Thanks a lot.

supermerill, 2021-12-01:
I'll modify the tooltip

```
This parameter grows the top/bottom/solid layers by the specified mm to anchor them into the **sparse infill**. Put 0 to deactivate it. Can be a % of the width of the perimeters.
```

What do you think?

Fabface, 2021-12-01:
That sounds good - now that I know what it really does the current tooltip does make sense as well, but I would have never understood it just from the tooltip alone.
Maybe write it like this:
_This parameter grows the top/bottom/solid layers by the specified mm to anchor them into the **sparse infill and support the perimeters above**. Put 0 to deactivate it. Can be a % of the width of the perimeters._

closed 2021-12-21 by Fabface