supermerill/SuperSlicer #1982: Volumetric flow rate seems stuck

nuxy69, 2021-11-29:
**Describe the bug**
Even though everything is on automatic, the volumetric flow rate seems to be stuck at 10mm3/s

**To Reproduce**
Steps to reproduce the behavior:
1. Set all the speeds to 0
2. Set the volumetric flow rate on print settings and filament to 30

**Expected behavior**
Volumetric flow rate should be 30 or close to it.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.57.6

neophyl, 2021-11-30:
Can you attach your saved 3mf project file (file>save project as) please.

nuxy69, 2021-11-30:
Yes, ofcourse.. Sorry about that
[Front_lead_screw_alignment_tool_v2.zip](https://github.com/supermerill/SuperSlicer/files/7624043/Front_lead_screw_alignment_tool_v2.zip)
t.

neophyl, 2021-11-30:
Don't set Gapfill speed to 0.  It doesn't work like the rest of the auto speeds. it even mentions it in the tooltip.  Set it to something reasonable like 20 or 30.  Doesn't really matter you can set it to 200 as long as its not 0.

Its always been an oddity.  It used to be setting it to zero was the only way of turning off gapfill.  Then a check box was added to turn it on/off.  At that point it was found that setting zero for its speed messed up the auto speed side of things.  That's why the tooltip specifically mentions setting to Zero will make it part of the autospeed calculation.

nuxy69, 2021-11-30:
Ah, still the same behavior as PrusaSlicer then. Didn't see the tooltip as I imported all the settings. 
Will close this issue. Thanks @neophyl

closed 2021-11-30 by nuxy69