supermerill/SuperSlicer #1983: "Invalid value provided for parameter fill_pattern:" when opening 3mf

numanair, 2021-11-30:
**Describe the bug**
Attempting to open a 3mf with a layer height-range modifier for infill results in an error and no file is opened.
Editing the Prusa_Slicer_layer_config_ranges.xml to add the missing infill type option lets Superslicer open the 3mf successfully.

> Failed loading file "problem.3mf" due to an invalid configuration.
> Invalid value provided for parameter fill_pattern: 

Prusa_Slicer_layer_config_ranges.xml contains a line `<option opt_key="fill_pattern"/>`
It should be more like `<option opt_key="fill_pattern">rectilinear</option>`

**To Reproduce**
I have not found exact steps yet. I know I had a file with this problem in which I added multiple height range modifiers with modification to infill type/density.

**>> Project File <<**
[range-modifier-problem.zip](https://github.com/supermerill/SuperSlicer/files/7623358/range-modifier-problem.zip)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.57.6

csdstudio78, 2022-02-02:
I think this may occur when copy/pasting (ie duplicating) objects with height layer mods.

supermerill, 2022-02-02:
If you can create a bad .3mf from a an empty platter, tel me how so I can reproduce it. Copy-pasting object with layer range isn't enough for me to make it bug.

csdstudio78, 2022-02-02:
OK... I'll see what I can find!  This corrupts my files weekly, I just wish I could find the exact steps to reproduce. :)  It only happens with files that have more than one instance of an object, so that's my only pointer to it being a copy/paste bug.

TPSSCW, 2022-03-06:
im having the same issue with a file i just created and saved yesterday, just started using SuperSlicer and love the features but not being able to open a saved file is not cool!

supermerill, 2022-03-06:
Next version: I integrated a fix to be able to open them, even if the fill_pattern can't be recovered.
I also added a warning message.
And it creates a log if it fails to create the project file properly so you can send it to me, so I can have a kind of clue.

TPSSCW, 2022-03-07:
is that in the curent beta version or in the next release?

supermerill, 2022-03-07:
the one i released just now

Dxebi, 2022-04-26:
Here my 3mf file where I get this error on version 2.3.57.12
[Duplo2Brio total.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8566232/Duplo2Brio.total.3mf.zip)

supermerill, 2022-04-26:
> Here my 3mf file where I get this error on version 2.3.57.12*

yes, it's the stable one, not the beta. The current beta is the 2.4.
And i still don't know why it's not printing the field value.

here with "trinagles in the two empty fields.
[1983_Duplo2Brio total.zip](https://github.com/supermerill/SuperSlicer/files/8566674/1983_Duplo2Brio.total.zip)