supermerill/SuperSlicer #1987: Repair STL then export, still needs repair?

chimeranzl, 2021-12-01:
If I add an STL (one that requires repairing) it says it auto-repairs it.  All good.  So I then export the STL file to effectively export the bed as a 'repaired STL', but when I clear the plate, then add the repaired STL back in, it says it's been autorepaired again.  It's like the export function doesn't export the STL with the repairs.  This used to work on an earlier revision (i think 2.2.something)

Running Superslicer 2.3.57.6 on Windows.

gpread, 2021-12-01:
I can confirm, this is the behaviour I have seen also.

Further more, while if I use the file>repair STL option which outputs as \*.obj and then re-import then the object is reported as clean.  If I then use the export as STL feature from the sidebar and then reload the STL it generates, it says it was repaired and that there were more errors than the original STL (2 errors becomes 60).

supermerill, 2021-12-01:
Yes, it's weird.
I read that prusa has modified their repair / store / load process in the 2.4, so hopefully it will be better after merging the changes.

gpread, 2021-12-01:
It was definitely working ok in earlier Susi 2.3.57.x and/or late 2.3.56.x versions

supermerill, 2021-12-01:
If you can pinpoint he version where it start to behave weirdly, and a way (with files) on how to reproduce, then I may be able to see what's the problem.
But I didn't touch the repair/import part in the .57 branch.

chimeranzl, 2021-12-01:
> Yes, it's weird. I read that prusa has modified their repair / store / load process in the 2.4, so hopefully it will be better after merging the changes.

Just downloaded and tested PrusaSlicer 2.4.0-beta2 and it works fine in that release.

supermerill, 2021-12-02:
The "repair" use a Microsoft api (ABI::Windows::Graphics::Printing3D) that use a cloud service (Netfabb?) to repair the stl. So the result may varies independently.

supermerill, 2021-12-02:
> In Susi (at least since 2.3.56.8) there is no "fix through netfabb", is that intentional and therefore something that will remain after Susi merges with PS 2.4.x ?

probably something that has been cut by the merge. I'll re-add it.

neophyl, 2021-12-02:
Not sure if this is the cause but recently over on the prusa forum someone mentioned that the netfabb service was not available like it used to be.  I know windows installs use some form of programmatic  based calls for it but I'm wondering if Autodesk have made other changes that could be causing it.
https://service.netfabb.com/ - We have completely redesigned the Netfabb Online Repair Service and merged it into
Autodesk Fusion 360.

supermerill, 2021-12-03:
Also microsoft now has a cloud repair servoce on azure (https://tools3d.azurewebsites.net/)

gpread, 2022-01-21:
> > In Susi (at least since 2.3.56.8) there is no "fix through netfabb", is that intentional and therefore something that will remain after Susi merges with PS 2.4.x ?
> 
> probably something that has been cut by the merge. I'll re-add it.
> 
> edit: seems like it's a compilation issue, as it's an optional dependencies that I have to manually add/set.

I hope you will be able to restore this PS functionality this with the 2.4 merge as its very useful, having to keep using PS just to fix dodgy stl files is a PITA