supermerill/SuperSlicer #1989: First layer  thin wall flow rate too many low about 100x low than perimeter

navr32, 2021-12-01:
I want to print thin wall for geometric logo design.
The thin wall are take in by superslicer with the Thin wall option enable  "Print Settings -> Perimeters & Shell-> Quality -> thin Walls check enable min width 30%  overlap 50% merging with perimeters enable" 

**To Reproduce**
I have try many many setting i have always the same problem.

Normally the flow for the first layer of my thin-wall must be about the same with about 10..20% of flow up than upper layer to have
a good adhesion to plate..

**Screenshots**
ON the Screenshot i have set the View to Volumetric flow rate .so you can see that the first thin wall layer is in dark blue 0.008 mm3/s flow...oupps...

**Desktop (please complete the following information):**
 Linux manjaro Gnome.  SuperSlicer 2.3.5
[flower_of_life_super_slice_problem.zip](https://github.com/supermerill/SuperSlicer/files/7636569/flower_of_life_super_slice_problem.zip)
7
[image]

Many Thanks

supermerill, 2021-12-02:
Can't reproduce on the 2.3.57.6
[image]

what is your version in help -> about?

navr32, 2021-12-03:
2.3.57.5

navr32, 2021-12-03:
But i see your test and you have the first layer with too low flow rate too ? it is better yes but not enought ! 
In prusa i have well the first layer more flow than the upper... here on prusa the first layer is well green so about 1.6 to 2 flow
and the upper about 1.2..1.6
[image]

supermerill, 2021-12-03:
You have a first layer speed of 10mm/s, that's why it's slower than for other layers that have a 60mm/s

navr32, 2021-12-03:
Perhaps adding one settings  in "Printes settings -> Width & Flow ->  Thin wall first layer"  ??? to be able to force the flow rate just for thin wall first layer.. like  infill..top infill and so... ?

navr32, 2021-12-03:
Yes but if i want to print slowly the first layer this flow is too low and I am unable to force the ratio at this speed...but it is possible for the perimeters ...witch one i am able to put more flow than upper even with 10mm/s flow..
[image]

supermerill, 2021-12-03:
0.3 height * 0.4 with * 10mm/s => 1.2mm3/s
I think that the color can be deceiving, i think that shade of blue means '1.2' (humans aren't good to distinguish hue, much better at brightness). 
I tried to show in the gui the value from the current nozzle position, but it's a bit complicated to do.

navr32, 2021-12-03:
I want to have the flow rate for the first layer thin wall to be more than the upper layer to have good bed adhesion...
Because you see that for perimeter the layer is green and upper blue mix green that's say the perimeter first layer have more flow than the first layer perimeter.... And i want to have the same for the thin wall first layers..have more flow for the first layer thin wall than the upper thin wall layer

navr32, 2021-12-03:
The simple will be to be able to force ratio flow for firt layer thin wall too ...like for perimeters ?

supermerill, 2021-12-03:
That's because you have a 260% width on the first layer.

what you can do: 
* set your first layer speed to 100% (same speed as other layers)
* set your max volumtric speed (in filament tab) to ~4mm3/s

navr32, 2021-12-03:
Yes this make sense....i was thinking that first layer width&flow->extrusion width -> first layer do the job....so it done the job here but just for perimeter....so i try your trick..

navr32, 2021-12-03:
Oups scuse me I have set the setting you tell me  but i have the first layer too low flow now...how to set a fat gapfill ? for all the first layer ?

navr32, 2021-12-03:
I  think that's the "first layer with" setting  work only on first layer perimeter but not on thin wall first layer...why ?  If i set it more  i have a good color variation for the perimeter but any change for first layer thin wall..

supermerill, 2021-12-03:
> .why ?
because the geometry isn't wide enough to allow that.