supermerill/SuperSlicer #1990: Adding similar function of SPEEDUPTIME for cooling but for Printing SPEED: slowdown in overhangs proximity to avoid curling.

Frasta37, 2021-12-01:
OVERHANGS CURLING
Slowing down printing speed for overhangs helps a lot to avoid curling but the slicer starts at that slower speed too late to avoid curling at the beginning of the overhangs.  IE: Internal perimeter speed is 80 - Overhang is 20, between the 2 there's no middle way to I'd like to do 80 --> slow down to 30 just at the end of perimeter for 1-2cm of extrusion ---> start at 20 the overhangs. Is that possible? it'd resolve the problem without compromising the overall printing speed 

SOLUTION
So my idea is to add a similar function of SPEEDUPTIME used in the cooling section of the printer to slow down the print speed before the start of the overhang.

HOW IT WOULD WORK
IE: Internal perimeter speed is 80 - Overhang is 20, between the 2 there's no middle way to I'd like to do 80 --> slow down to 30 just at the end of perimeter for 1-2cm  ---> start at 20 the overhangs. Is that possible? it'd resolve the problem without compromising the overall printing speed 

Is it something possible to add? 

Thanks

supermerill, 2021-12-02:
> Is it something possible to add?

not now, as I need to finish a better api for post-processes.

The issue is caused by a lack of pressure advance, so when it slow down before the overhang, the pressure won't do the same?

Frasta37, 2021-12-02:
The idea it's just to give more time to the overhangs to cool down properly.. starting the overhang speed a couple of cm in advance like the speedup fan settings. Thanks anyway.. no stress.. but I'd love to have it in the future :)