supermerill/SuperSlicer #1991: Label bug in Custom Printer Settings

fmeroney, 2021-12-01:
**Describe the bug**
When setting up a custom printer I went to update the “Printer Settings” under the tab. Under Machine Limits, The default settings under Max Feedrates seem to be very low for the “Label” mm/min. 
- Max Feed X: 500 mm/min 
- Max Feed Y: 500 mm/min
- Max Feed Z: 12 mm/min
- Max Feed E: 120 mm/min

the speed rates seem to be mislabeled and should be mm/Sec OR if the label is correct then default values are very low.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ‘config assistant’
2. Click on ‘Custom Printer, create new printer'
3. Go to ‘Machine Limits’
4. Under ‘Max Feedrates’ 
5. See Default Values Vs. Label mm/min

**>> Project File <<**
NONE

**Expected behavior**
1. Label should read mm/Sec
OR
2. If label is correct, default values are misleading. 

**Screenshots**

**Desktop (please complete the following information):**
 - OS: [MacOSX]
 - Version [e.g. 2.3.57.6], Use `Help->About`_

**Additional context**
NONE

fmeroney, 2021-12-01:
[image]

neophyl, 2021-12-01:
I'm guessing you have a gcode flavour like repetier selected ?   If you select Marlin the feedrates are listed in mm/s.  However the default values dont change as they are just placeholders.  
The whole point of setting up a custom printer is that you should know enough to set those correctly for your printer.  As such they could all be set to zero and be just as valid.

I suspect the defaults were created with Marlin in mind.  I also vaguely remember github issues being raised about the units label being incorrect for certain types of gcode flavour and didnt match the firmware type.  So iirc the labels get changed now depending on type selected.  But as mentioned previously the placeholder values don't.

fmeroney, 2021-12-02:
It’s Klipper, I assume the values are in mm/sec. 

- Frank

supermerill, 2021-12-02:
I tried to change the unit by firmware type.
I think it's a mistake, s now you can't change the firmware selection without rewritign evrythign (or a bit of code to do taht automatically)
I'll have to revert to a fixed mm/s for everything and a convert at the export stage.