supermerill/SuperSlicer #1992: layer heights problem

legend069, 2021-12-02:
### Version
_Version of SuperSlicer used goes here (help->about)_
Version 2.3.57

### Operating system type + version
_What OS are you using, and state any version #s_
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition

### Behavior
* _Describe the problem_ layer height is adjusted by bridge flow ratio
* _Steps needed to reproduce the problem_adjusting bridge flow ratio then looking at the layer height
* _Expected Results_ it to not change 
* _Actual Results_ layer height changes. messes with other parts of print if using adaptive layers
  * _Screenshots if useful_

_Is this a new feature request?_ no
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[image]
[image]
[image]
[image]
[slicer_config_and_gcode_file_used.zip](https://github.com/supermerill/SuperSlicer/files/7642616/slicer_config_and_gcode_file_used.zip)

_

supermerill, 2021-12-02:
I think you forgot to replace the template text by your own.
There is no description to read. I don't really understand what's wrong here.

supermerill, 2021-12-02:
Please save your project file(s) : File -> save project as
It's mucgh much easier for me.

legend069, 2021-12-02:
> Please save your project file(s) : File -> save project as
> It's much much easier for me.

Sorry I thought I did that.
I'll do that in the morning

legend069, 2021-12-03:
I've added the file into the zipped folder

[slicer_config_and_gcode_file_used.zip](https://github.com/supermerill/SuperSlicer/files/7647556/slicer_config_and_gcode_file_used.zip)

supermerill, 2021-12-03:
Can't spot anything wrong
Changing the flow ratio of the bridge modify the height & width of it, and nothing else.

legend069, 2021-12-03:
> o of the bridge

so if i have " flow ratio of the bridge" set to 80%
and adaptive layers enabled  this would be changing z height to compensate for the bridge area that is 0.360 and then back down to 0.100 correct?

then for the travel movements across that area  z height change ?

hopefully that makes sense

[image]

supermerill, 2021-12-03:
sorry i don't understand.

legend069, 2021-12-03:
> sorry i don't understand.

in attached screenshots, the nozzle moves overtop of the area that is a different layer height but the z position doesn't go up to go over the height layer

[image]
[image]

neophyl, 2021-12-03:
Merill I think they are having issue with the fact that for bridging the layer height is equal to nozzle width and not 0.1 like the rest of the layers.
What they (and most people it seems) don't understand that ANYTHING printed as a bridge is not squished against the layer below and so its modeled as a height=width as that's how the plastic comes out the nozzle.  I think they were expecting the bridge layers to be printed higher up as with a thicker height to make the bottom match the model they would need to be.  

I think the constant stream of issues relating to this function was what finally made Prusa have the option to change this behavior in their 2.4 builds.  I must admit in all the printing I've done I always turned that off to get rid of the bridge flow wherever possible as it always printed better without.  

Basically most of these issue will go away after the 2.4 merge and all we will be doing after that is constantly pointing out which setting to change to use the thinner bridge height.

supermerill, 2021-12-03:
I was also thinking about moving the bridge layer a bit higher to be sure it doesn't overflow on the empty area... later.

@legend069 sadely, no, that's how it works. You can reduce the bridge flow to reduce the overflow.

celtare21, 2021-12-03:
> I was also thinking about moving the bridge layer a bit higher to be sure it doesn't overflow on the empty area... later. btw, the "bridge kill switch" is just a way to lie. It won't stop physic to create a cylinder because there is nothing under it.
> 
> @legend069 sadely, no, that's how it works. You can reduce the bridge flow to reduce the overflow.

Would asking for a port of the "thick_bridges" option from prusaslicer 2.4 be too much? Or you'd rather not mess with it as i heard there's plans to update the prusaslicer version at some point.

supermerill, 2021-12-03:
> Would asking for a port of the "thick_bridges" option from prusaslicer 2.4 be too much? Or you'd rather not mess with it as i heard there's plans to update the prusaslicer version at some point.

There's plans to update the prusaslicer version when out of beta.

also, disbaling "thick bridge" is the same as 'please i want to ignore the physics and just want a pretty preview'.
It's the same as reducing the bridge flow anyway. I'll add some user-friendly way of tuning the flow at the same time.

celtare21, 2021-12-03:
> > Would asking for a port of the "thick_bridges" option from prusaslicer 2.4 be too much? Or you'd rather not mess with it as i heard there's plans to update the prusaslicer version at some point.
> 
> There's plans to update the prusaslicer version when out of beta.
> 
> also, disabling "thick bridge" is the same as 'please i want to ignore the physics and just want a pretty preview'. It's the same as reducing the bridge flow anyway. I'll add some more user-friendly way of tuning the flow at the same time.

Well, for me it's not about a "pretty preview". I find that "thick bridges" are detrimental to any kind of bridge or overhang area. I understand the theoretical reasoning it is there, but at least in my opinion in practice it is quite the opposite. Bridges printed at 0.2mm look beatiful while bridges printed "thick" are way thicker than they need to be and it really shows not only making it ugly, but if this was a hole for example printed vertically, the "thick" also totally messes the dimensions and usually requires post processing.

supermerill, 2021-12-03:
then reduce the bridge flow and decrease the bridge overlap to add some spacing. It's the same result if you enter the good values.

celtare21, 2021-12-03:
> then reduce the bridge flow and decrease the bridge overlap to add some spacing. It's the same result if you enter the good values.

That also changes the line width, which is not wanted. I don't really see any way to replicate "thick_bridges" toggle in susi as of now.

supermerill, 2021-12-03:
The line width is purly virtual, the gcode contains only flow.

celtare21, 2021-12-03:
Oh, I didn't know that at all, thank you for the explication

legend069, 2021-12-14:
thanks for clearing it up
@supermerill

closed 2021-12-14 by legend069