supermerill/SuperSlicer #1993: Top layer fan speed issue - Oscillating fan - Default fan speed during retraction moves

GutSmudge, 2021-12-02:
2.3.57.6 Version 
_Version of SuperSlicer used goes here (help->about)_

Windows 10 Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
Top fan speed is "only" applied when **extruding** top layer. If a retraction occurs and e.g. a travel move is inserted, the Default fan speed will be used if "Run the fan at default speed when possible" is on. This is not so noticable with connected top infill, but when using rectelinear the fan will ramp up and down everytime a line end is reached. Also it does not apply the top fan speed when the ironing pass is done. This issue is especially troublesome with higher temp filament.
IMO "top fan speed" should completely override default fan speed when a top layer is printed - also during retraction. The increased cooling can mess up the ironing pass, especially using steel nozzles with worse thermal conductivity, as fan draws too much heat away causing tearing issues to the surface.

* _Steps needed to reproduce the problem_
Slice a part with:
"Run the fan at default speed when possible" on
"Top fan speed" off (=1) or different from "Default fan speed"

View g-code or print part. Fan speed is default during retraction moves.

* _Expected Results_
No fan in entire top layer progress 

* _Actual Results_
Fan uses default fan speed during retration moves in top layer, sometimes causing oscilating fan or overcooling

  * _Screenshots if useful_
  * 
[image]

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
[Top_layer_fan_issue.zip](https://github.com/supermerill/SuperSlicer/files/7643917/Top_layer_fan_issue.zip)
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2021-12-02:
I tag that with "bug" to encourage me to fix it, but it's more an unwanted "feature" |-D
you can add a little 'fan speedup time'  and it should be as you want.

supermerill, 2021-12-11:
Currently, They're not a good way to resolve taht. have to wait for gcode writer rewirte.

ChaosBlades, 2022-10-03:
I have encountered this. If you have a kick start time on your part cooling fan. For example mine is 0.5 seconds then if you try to disable the part cooling fan with this it ends up actually increasing it since it will oscillate between 0% and 100% every second. It will even pile up in the queue and I have had the part cooling fan do this up to a minute after the print had finished. If I set it to 10% then it will oscillate between 10% and 50% on every line of the top layer (Monotonic Filled Not Connected). So it is only really working if you are printing a very large top layer at best and doing the exact opposite of what you are asking it to at worst.

This issue will also effect https://github.com/supermerill/SuperSlicer/issues/3271 if it gets implemented.

ChaosBlades, 2022-10-05:
Actually this effects all of these fan modes including External Perimeter fan speed because if it prints small pieces that are only 1 small external perimeter the fan will oscillate. Also you want fan speed on an external perimeter under a certain size.

I think in the mean time all the tooltips should mention this is not compatible if your fan has a `kick_start_time:`. I don't see a way this could be coded around unless these settings can only apply to things that are over a certain length.