supermerill/SuperSlicer #1995: Tools are not called in beginning of print.

terencemuscat, 2021-12-03:
**Describe the bug**
Missing Tool Call Number in Gcode at start of print. (ie T0 / T1)

**To Reproduce**
Steps to reproduce the behavior:
1) Print using default extruder
2) Tool call number is called if multiple extruders are utilized in a print. For example, having T0 & T1, in the same print will call both of them (T0 in the beginning and T1 at the toolchange point). However, having only T0 in the print will call nothing, so it will start printing without any tool.

supermerill, 2021-12-03:
[image]

terencemuscat, 2021-12-03:
Thanks, now if you try it with just one shape-box assigned to the default extruder, t0 will not be called.

supermerill, 2021-12-03:
If you need it, add it to your start-gcode
`T{current_extruder}`