supermerill/SuperSlicer #1998: Over-extrusion for nozzle diameter.

mafotaso, 2021-12-03:
### What happened?

Hello.

I ask for this problem a couple months ago and it was fixed more or less but now i would like solve it completely.

It is about over-extrusion and stuck problem on some pieces when the Superslicer want extrude more than 200% nozzle diameter, is there any parameter to say this maxim flow? I can´t find in this new 2.3.57.6 version neither.
[image]

Because of that I have stucks sometimes and waste a lot plastic or visual defects.

### Project file & How to reproduce

[overextrusion.zip](https://github.com/supermerill/SuperSlicer/files/7650766/overextrusion.zip)

### Version

2.3.57.6

### Operating system

win 10 version 21H1

### Printer model

custom, near of prusa printer

supermerill, 2021-12-03:
remove gapfill, and replace all rectilinear+gapfill by rectilinear.

mafotaso, 2021-12-03:
> 

Thanks for your fast reply. 

I already have activate "overlapping perimeter" and "for all perimeters". That was the improve months ago for the same question.

Where I can find  "replace all rectilinear+gapfill by rectilinear."  I have found deactivate gapfill, (but appear the same over-extrusion 0.86mm) but I don't know where is rectilinear+gapfill sorry.

supermerill, 2021-12-03:
yes