supermerill/SuperSlicer #1999: Option to generate Filament Card

schulzh, 2021-12-03:
It would be great to have the ability to generate a filament card, e.g. like this one: https://www.thingiverse.com/thing:4274329 from the slicer. The slicer could then emboss the filament name and settings into the card itself, removing the need for printing/writing a label or to create the emboss manually.

supermerill, 2021-12-03:
* download [freecad 0.19]() and isntall it
* add the free cad path to your preference (often C:\Program Files\FreeCAD 0.19 on window)
* add the fillamentSample and put the origin where you want the text to start
* open the Generate ->freecad python script
* type
```
cut("FilamentSampleWithName")(
	$1,
	linear_extrude(10)(
		text("redish-thing pla", 5),
	).move(z=1.4)
),
```
* click generate (you can select "insert instead of replace the first times, so you can test more easily)
[image]

supermerill, 2021-12-03:
I don't have placeholder for variables. I'll see if i can add them into the python.