supermerill/SuperSlicer #2005: Broken URLS for Calibration Cube

NikosSprocket, 2021-12-04:
### What happened?

For "...made by iDg3Dprining"
15:09:53: Could not set current working directory (error 123: The filename, directory name, or volume label syntax is incorrect.)
15:09:53: Unable to open requested HTML document: https://www.thingiverse.com/thing:1278865

For "....made by vorondesign"
Broken URL.

### Project file & How to reproduce

SuperSlicer_2.3.57.6_win64_211118

### Version

Version 2.3.57

### Operating system

Windows 10 64bits

### Printer model

Voron 2.4

supermerill, 2021-12-06:
I don't understand

louis-lau, 2021-12-07:
This may be a bit clearer:

https://user-images.githubusercontent.com/1346804/145026728-54672306-4453-488e-badc-15bea7dfcfcb.mp4

NikosSprocket, 2021-12-07:
Exactly as @louis-lau showed with his video captured clip.

Thanks!

closed 2021-12-22 by supermerill