supermerill/SuperSlicer #2006: Insanely high flow rates

joaoferreira-git, 2021-12-04:
### What happened?

When i slice anything it just gives insane flow rates and almost made my extruder self destruct, it happens with included profiles and also with the ones i tweaked.

[image]

### Project file & How to reproduce

Slice the model and then select view type to volumetric flow rate.

[project-files.zip](https://github.com/supermerill/SuperSlicer/files/7654943/project-files.zip)

### Version

2.3.57.6

### Operating system

Windows 10

### Printer model

Voxelab Aquila

supermerill, 2021-12-06:
you set 96.77 in your extrusion multipler.
it'a multiplier, not a percentage.
that means a flow of 9677%

joaoferreira-git, 2021-12-07:
Im sorry for the report i also figured it out today, i was going to delete the bug report now.

closed 2021-12-07 by joaoferreira-git