supermerill/SuperSlicer #2007: Fat Benchy Chimney

SturmB, 2021-12-05:
### What happened?

**Problem:**
I have a Prusa i3 MK3S+ and I'm having a weird issue with the chimney of the #3DBenchy. This is printed from SuperSlicer v2.3.57.6 with Prusament PLA Galaxy Black at 215℃ nozzle temp., 60℃ bed temp., 0.15mm layer height, and the stock 0.4mm nozzle. All other settings are at their defaults. I even tried lowering the nozzle temp. to 200℃ and got the same result.

Watching it extrude this, it appears to be "pushing" the already-extruded plastic outward as it lays down the next layer. The extrusion modifier is at its default of 1.0, so I have serious doubts that it's overextruding. I have also tried this with three different Prusament PLAs with the exact same results. In addition to the messed-up chimney, an awful grinding sound can be heard when the extruder slows down. (See videos below.) I cannot tell from where the sound is coming, nor do I know how to localize it.

Printing the same thing from PrusaSlicer with the same settings, however, results in a perfectly clean chimney and no awful grinding sound during printing.

**Expectation:**
For the chimney to print normally and without a grinding sound.

**Evidence:**
Please see the following two YouTube videos for comparison. You will want to turn up the volume so you can hear the awful grinding sound during the SuperSlicer video:

* [SuperSlicer](https://www.youtube.com/watch?v=LPSb7aduHwA)
* [PrusaSlicer](https://www.youtube.com/watch?v=T9VK8dCixQ8)

And here's a close-up image of the benchy as a whole with the messed-up chimney, printed in Prusament PLA Lipstick Red. You can probably tell that the rest of the benchy is quite passable:

[image]

### Project file & How to reproduce

I am including both `.3mf` files as well as their `.gcode` counterparts. I sincerely hope they help figure out from where the problems are coming:

[benchy_chimney.zip](https://github.com/supermerill/SuperSlicer/files/7654977/benchy_chimney.zip)

### Version

2.3.57.6

### Operating system

Windows 10 20H2 (19042.1348)

### Printer model

Prusa i3 MK3S+ with normal E3D heatbreak

supermerill, 2021-12-06:
There is too much segment for the gapfill, and that overload your printer firmware.
set your "max G1 per second" to 150.

I think i'll link the gapfill resolution to the resolution setting and 'reduce' the resolution to 0.0125 to be in line with PS by default.

SturmB, 2021-12-06:
Indeed, that seemed to alleviate the issue. Thanks, @supermerill! It does still make that 'grinding' sound a little during retraction—I'm guessing at the end of a layer—when the whole extruder slows down almost to a stop. But the sound is nowhere near as bad as it was.

The only visual difference I see now between a PS-sliced chimney and SS-sliced chimney (keeping all settings at their defaults) is the top:

[image]

The chimney on the right, using PrusaSlicer, looks arguably better than SuperSlicer's result (on the left) because of one other setting difference (that I could find). Here's how it looks in PrusaSlicer:

[image]

And here's how it appears in SuperSlicer:

[image]

It seems the setting **Print Settings** -> **Perimeters & Shell** -> **Quality** -> **Only one perimeter** -> **On top surfaces** is checked by default, which adds that top layer. And, indeed, when I un-check that setting, the top looks identical to PrusaSlicer's (much better).

So, I guess for something as small as this chimney, apparently, this setting is detrimental to its final appearance. I wonder how PrusaSlicer knows when to use a top layer and when not to…

neophyl, 2021-12-06:
In PS there isn't any room for a top layer as the perimeters take up thew whole space.  With the single perimeter on top option there is enough room.  Which is an option that PS doesn't have at all.  So its not a case of 'knowing'.

Id be checking if you have a slight bit of over extrusion, even something as small as not accurising your filament diameter to match reality could cause it.  For example if your filament is 1.76 instead of 1.75.  Most people play around with extrusion multipliers and forget that all the math for extruded plastic volume is based on the filament diameter.  If there is a difference between reality and the assumption of 1.75 then it can and does cause over/under extrusion.

SturmB, 2021-12-06:
> Id be checking if you have a slight bit of over extrusion

I did run SuperSlicer's **Filament Flow Calibration** recently with this exact spool of red Prusament and came up with an Extrusion Multiplier of 1.034. However, as I have stated before, all the photos and videos I've taken for comparison purposes are with all settings at their **defaults**—including an Extrusion Multiplier of 1.0.

That said, it could be that my spool of Prusament is 1.76mm or 1.77mm. Either way, the Filament Flow Calibration gave me an EM of 1.034. So either the calibration is flawed or I'm just not following its instructions correctly. But now we're straying from the original topic anyway.

supermerill, 2021-12-06:
> now we're straying from the original topic anyway.

we try to resolve all the extrusions problem from the chimney.

If you print abs, I may advise to decrease a bit the filament 'max line overlap' (to something like 40%), as abs has some difficulties flowing under the next extrusion.

macchinette, 2021-12-06:
> There is too much segment for the gapfill, and that overload your printer firmware. set your "max G1 per second" to 150.
> 
> I think i'll link the gapfill resolution to the resolution setting and 'reduce' the resolution to 0.0125 to be in line with PS by default. As this creates a trap.

I wanted to contribute if I can by saying that the command "max G1 per second" would be correct by default at 150 and not at 1500 as it is currently, since the base of the boards is 8bit.

Then with the resolution (0.0001 or 0.0125) different results are obtained for the alignment of the seams, practically in different points such as the Benchy if you slice at 0.0001 resolution there are different effects compared to 0.0125 for the seam positioning it at the rear of the boat.
A separate double setting of "Rear" and "Aligned" would almost be needed.

Resolution 0.0001 worse everything and better the fireplace

[image]
Resolution 0.0125 better everything and worse the fireplace

[image]

SturmB, 2021-12-06:
> If you print abs, I may advise to decrease a bit the filament 'max line overlap' (to something like 40%), as abs has some difficulties flowing under the next extrusion.

I'm printing PLA currently, as noted in my original post. However, I did make the overlap adjustment you suggest anyway and it does appear to result in a slightly cleaner top surface. Thank you again! 😀