supermerill/SuperSlicer #2008: Arrange Bug with Hidden Objects

littlejon, 2021-12-05:
### What happened?

With multiple items on the build plate, if any are marked as "hidden" the Arrange feature places both the hidden item and the visible item outside of the print area (screenshot and 3mf file attached).

### Project file & How to reproduce

![Auto Arrange with Hidden Object](https://user-images.githubusercontent.com/955638/144735899-da4376f4-a1fb-4557-94ae-0f531c486f99.png
[Auto Arrange with Hidden Object.zip](https://github.com/supermerill/SuperSlicer/files/7655319/Auto.Arrange.with.Hidden.Object.zip)

### Version

2.3.57.6

### Operating system

Windows 11 (22000.318)

### Printer model

Ender 3 Pro & Ratrig v-core 3

neophyl, 2021-12-05:
Turn OFF Auto Center parts in Preferences.

littlejon, 2021-12-05:
@neophyl - thank-you. Disabling the "Auto-center Parts" options gives the expected behaviour.

Is this an issue with the Auto Center parts functionality or am I missing the point?

littlejon, 2021-12-05:
Ok, I get it now.

I see how this could be useful, but it strikes me as an unhelpful default option.

neophyl, 2021-12-05:
Its the same in Prusa Slicer.  Agreed on the default, there are many that dont make sense but probably did/do for certain use cases or at the time they were added.
One of the first things I do is turn off the auto center.  I like to move parts around the bed so they aren't all printed in the middle.  Besides loading a new object puts it to the center by default (3mf being the exception at least in builds based on PS2.3)

supermerill, 2021-12-06:
I think the issue is that "hiding" a part doesn't trigger the auto-center, and un-hidding it does.

closed 2021-12-22 by supermerill