supermerill/SuperSlicer #2009: Overhang/Bridge too slow due to flow rate limit

LeLocTai, 2021-12-05:
### What happened?

I use a rather big nozzle, so I make use of the "Volumetric speed for Autospeed" setting to limit flow rate on fat gap fill.

Due to the way bridge's layer height is always = nozzle size, their calculated volume when using big nozzle is much greater than the actual extruded volume. This cause the bridge speed to be pulled down way too low.

Volumetric speed for Autospeed = 13
[image]

Volumetric speed for Autospeed = 0 // Disabled
[image]

### Project file & How to reproduce

[Bridge Problem.zip](https://github.com/supermerill/SuperSlicer/files/7655402/Bridge.Problem.zip)

### Version

2.3.57.6

### Operating system

win10

supermerill, 2021-12-06:
I don't see how this can be a bug.
If you want lower bridge mm3/mm, then lower the birdge flow.
If you want higher vol flow than the max vol flow (from autospeed), then set the desired speed yourself.

Is there a third or fourth option?

LeLocTai, 2021-12-06:
The problem currently is autospeed is limiting the speed of a bridge move but not any other moves of the same flow. You can see it in the screenshots: The bridge speed (60) is slower than both perimeter and infill speed, but with auto speed on it is limited all the way down to ~30. Bridge flow is less than that of other move.

I'm have not looked at the code so I can't be sure, this is only a guess. Let say bridge flow is 100% for simplicity. The volume of a non bridge would be `length*nozzle_size*layer_height`. Bridge seem to use `length*nozzle_size*nozzle_size`. So with large nozzle size, bridge is limited to very slow.
This is same as the first image, auto speed is on but nozzle size is reduced:
[image]

To show that bridge flow is not the culprit, I set vol flow to very low. With .4 nozzle, every move have roughly the same speed:
[image]

By only changing to a .8 nozzle bridge is much slower than other:
[image]

supermerill, 2021-12-06:
> Bridge seem to use length*nozzle_size*nozzle_size. So with large nozzle size, bridge is limited to very slow.

bridge is length*nozzle_size*nozzle_size * PI/4 * bridgeflow
So yes, as your vol flow is limited, it has to slow down as the bridge has a bigger section.

LeLocTai, 2021-12-06:
Is that the actual extruded volume then? I was under the impression that the extruded amount depends on layer height. I suppose I can set bridge flow individually for each combination of layer height and nozzle I use, but I still think the better choice would be for bridge to have the same "layer height" as other move. Would be even more important when support is involved I think.

LeLocTai, 2021-12-06:
Ok so it looks like prusa called this thick bridge and it can be disabled in their 2.4. Guess I'll wait for the merge. Thanks a lot for your works here!

closed 2021-12-06 by LeLocTai