supermerill/SuperSlicer #2014: Placeholder bounding_box[0] not working

ding-bat, 2021-12-05:
### Version 2.3.57.6

### win11

Cant get bounding_box[0] to work, I put it in 'Start G-Code' but throws error on slicing
* _Steps needed to reproduce the problem_

In Printer -> 'Start G-Code' :
;klipper macro
;--------------------------------------------------------------
START_PRINT BED=[first_layer_bed_temperature] EXTRUDER=[first_layer_temperature] FLOW=[extrusion_multiplier] XMIN=[bounding_box[0]]
;--------------------------------------------------------------

mobilemcclintic, 2021-12-06:
Try {bounding_box[0]}

ding-bat, 2021-12-06:
Yes this works!

Thankyou for your help.

closed 2021-12-06 by ding-bat

ding-bat, 2021-12-06:
Just one thing I found.
bounding_box Array order to me is X_min / Y_min / X_max / Y_max.

like bottom corner / top corner.

;--------------------------------------------------------------
;klipper macro
START_PRINT BED={first_layer_bed_temperature} EXTRUDER={first_layer_temperature} FLOW={extrusion_multiplier} XMIN={bounding_box[0]} YMIN={bounding_box[1]} XMAX={bounding_box[2]} YMAX={bounding_box[3]}
;--------------------------------------------------------------

supermerill, 2021-12-06:
like said in https://github.com/supermerill/SuperSlicer/wiki/Variable-list, no?

ding-bat, 2021-12-06:
Yes variable list is good, #495 has it different way around, just a bit of confusion.

Anyway, looks like klipper doesn't do M557

12:51:51 
File opened:test_static_0.1_265_85_0.9.gcode Size:1172
12:51:51 
File selected
12:51:51 
// BED 85C, EXT 265C, FLOW 0%
12:51:51 
// XMIN 115, YMIN 105, XMAX 145, YMAX 155
13:01:44 
**// Unknown command:"M557"**
13:01:55 
// probe at 22.000,10.000 is z=1.130156

Will investigate.

Thanks for your help, again. :)

supermerill, 2021-12-06:
Oh, right.
I have to modify the 495 one, it's not consistent.