supermerill/SuperSlicer #2016: Crash on latest release shortly after startup

Melarok, 2021-12-06:
### What happened?

After updating from 2.3.57.5 to 2.3.57.6 SuperSlicer crashes shortly after startup. Terminal output if run from terminal:

### AUR:
```
15:03:00: Debug: Adding duplicate image handler for 'JPEG file'
15:03:00: Debug: Unrecognized accel key 'Dèl', accel string ignored.
15:03:00: Debug: "Shift+Tab" is not supported as a keyboard accelerator with GTK
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it
15:03:00: Debug: window wxTreeCtrl(0x561b75a48240) lost focus even though it didn't have it

(superslicer:31977): Gtk-CRITICAL **: 15:03:09.195: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(superslicer:31977): Gtk-CRITICAL **: 15:03:09.195: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(superslicer:31977): Gtk-CRITICAL **: 15:03:09.195: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(superslicer:31977): Gtk-CRITICAL **: 15:03:09.195: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
**
Pango:ERROR:../pango/pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
Bail out! Pango:ERROR:../pango/pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
```

### AppImage:
```
(superslicer:32344): Gtk-CRITICAL **: 15:08:20.851: gtk_window_resize: assertion 'height > 0' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:20.964: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:20.964: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:20.964: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:20.964: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:20.964: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:20.964: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:20.964: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:20.964: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:21.166: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:21.166: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:21.166: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:21.166: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:21.167: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:21.167: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:21.167: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:21.167: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:32344): GLib-GObject-WARNING **: 15:08:21.167: invalid (NULL) pointer instance

(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:21.167: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[... 123 lines cut ...]
(superslicer:32344): GLib-GObject-CRITICAL **: 15:08:21.184: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
**
Pango:ERROR:../pango/pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
Bail out! Pango:ERROR:../pango/pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
```

### Project file & How to reproduce

Start SuperSlicer
Go to custom G-Code section for the printer
Copy stuff from there
After an inconsistent time it crashes
[Crash.zip](https://github.com/supermerill/SuperSlicer/files/7661401/Crash.zip)

### Version

2.3.57.6-2 compiled from the AUR and 2.3.57.6 as AppImage

### Operating system

Manjaro KDE Edition - testing branch

### Printer model

Ender 5

supermerill, 2021-12-06:
Did you try the gtk2 build?

Melarok, 2021-12-06:
I just tried it and it crashes with the exact same pango error.
Actually the whole terminal output is the same (all those GLib Warnings)

neophyl, 2021-12-06:
Might want to look at this issue on Prusa Slicer https://github.com/prusa3d/PrusaSlicer/issues/7431  (currently been closed)

supermerill, 2021-12-06:
thanks @neophyl, because i didn't touch any gui thing between .5 and .6

closed 2021-12-06 by Melarok