supermerill/SuperSlicer #2017: Decreased quality

macchinette, 2021-12-06:
### What happened?

I have only now noticed that the surface quality has deteriorated, I think it is already present in previous versions, but I don't know which version it started from.
I made the comparison with PrusaSlicer beta3 with the same settings and the result is much better, as it was SuperSlicer long ago.

It does not seem to me that there are parameters that conflict with each other.

If it is useful I can try previous versions to identify which version started this behavior from.

SuperSlicer
[image]

PrusaSlicer
[image]

### Project file & How to reproduce

[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/7663831/3DBenchy.zip)

### Version

Version: 2.3.57.7 Build: SuperSlicer_2.3.57.7_2021-12-06

### Operating system

Windows 10

### Printer model

Anycubic I3 Mega S

supermerill, 2021-12-07:
that's coming from "min extrusion length". Before, it was not very efficient, and so needed a higher value than expected. Now it's fixed and it's very effective at fusing segments.
Now that there is "max g1 per second", i advise to don't use it.
Hint: check the difference between sliced preview and gcode preview (in tabbed layout). The sliced preview has the precision from the slicing, and you can see the degradation from the "max g1" or "min length".

macchinette, 2021-12-07:
> che proviene da "lunghezza minima di estrusione". Prima, non era molto efficiente, e quindi aveva bisogno di un valore più alto del previsto. Ora è fisso ed è molto efficace nel fondere i segmenti. Ora che c'è "max g1 al secondo", consiglio di non usarlo. Suggerimento: controlla la differenza tra l'anteprima affettata e l'anteprima gcode (nel layout a schede). L'anteprima affettata ha la precisione dello slicing e puoi vedere il degrado dal "max g1" o "min length".
> 
> Per ottenere lo stesso livello di dettagli, penso che sia anche necessario spingere il g1 massimo a ~ 750

Hi, I did some tests with the parameters you indicated, but I don't find the full benefit.
Let me explain, if I set "minimum extrusion length" to zero and to 750 "max g1 per second" I get the result below in the picture.
The problem is that the file size has doubled, but the quality of the G-CODE preview still has some flaws, while with any setting the sliced preview is perfect.
Now I'm wondering what are the right settings? Because some time ago I found myself with the printer that was not fluid and made ugly perimeters by printing circular objects.
Thanks
[image]

Pavulon87, 2021-12-08:
It looks more like a seam.
For circular objects, it is a good idea to use of [ArcWelder](https://github.com/FormerLurker/ArcWelderLib)