supermerill/SuperSlicer #2018: [FR] Add option for filament change gcode to go after first travel move on new layer

ManuelMcLure, 2021-12-07:
**Is your feature request related to a problem? Please describe.**
Usually, a new layer doesn't start at the same point the old layer finished - there's a X/Y travel move at the very beginning of the new layer just after the change in Z. The problem is that the `M600` gcode for a color change is inserted before this travel move. This means that layer N finished, Z changes to the height of layer N+1, and then the `M600` triggers. The printer moves to a park position, allows changing filaments, and then when the filament change is complete the printer moves back to the last position for layer N, often leaving a small bit of the new colored filament there. If this is on a top layer, there will be a noticeable blemish in the finished print.

**Describe the solution you'd like**
Have an option to place the color change gcode after the first travel move on the new layer. This might leave a small spot of the previous color at that location, but it should be covered up by the new color when that starts extruding.

**Describe how it would work**
Currently generated gcode example:
```
;LAYER_CHANGE
;Z:10.7
;HEIGHT:0.15
; LAYER: 70, Z HEIGHT: 10.7
G1 Z10.7 F7800
;COLOR_CHANGE,T0
M600
G92 E0
G1 E-0.7 F2400
G1 X62.169 Y97.508 F7800
G1 E0 F2400
```

Desired gcode:
```
;LAYER_CHANGE
;Z:10.7
;HEIGHT:0.15
; LAYER: 70, Z HEIGHT: 10.7
G1 Z10.7 F7800
;COLOR_CHANGE,T0
G92 E0
G1 E-0.7 F2400
G1 X62.169 Y97.508 F7800
M600
G1 E0 F2400
```

The `M600` is after the retract, after the first travel move, and before the unretract.

Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
Currently I use manual editing to change the location of the `M600`. It's possible that there may be a way to use a postprocessing script to move the `M600` after the first travel move beyond the original location of the `M600`

supermerill, 2021-12-07:
current & desired are the same?

ManuelMcLure, 2021-12-07:
No - look at the position of the `M600` - the current is just after the `COLOR CHANGE` comment, the desired is three lines down (after the `G1 X62.169...`)

ManuelMcLure, 2021-12-07:
I've updated the gcode to move the comment as well as the `M600` - it makes more sense that way.

Triptikal, 2022-09-29:
Hello, 
I found out the same problem. Seems SuperSlicer puts the M600 command "in the middle of " the layer change instructions causing the printhead to print this layer "in the air" ( because in my case I'm adding 80mm to the Z to move the printhead in a "safe" place to execute the filament change )  before the next instruction of layer is given to the printer. It is resolved by manually move the M600 command just before the layer change instruction.
The enhancement should be , from my pov, that superslicer should insert the M600 command before the layer change instruction.

The generated code by superslicer : 
`G1 X142.009 Y143.673 E0.08266 ; Solid infill
G1 X144.28 Y141.403 E0.14565 ; Solid infill
;LAYER_CHANGE
;Z:1.32
;HEIGHT:0.28
; custom gcode: before_layer_gcode
;BEFORE_LAYER_CHANGE
;1.32
G92 E0
; custom gcode end: before_layer_gcode

;WIPE_START
G1 F6240
G1 X143.361 Y142.322 E-0.95 ; wipe and retract
;WIPE_END
G1 E-0.05 F4800 ; retract
G1 Z1.32 F7800 ; move to next layer (4)
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;1.32
; custom gcode end: layer_gcode

;COLOR_CHANGE,T0,#0ED51F
; custom gcode: color_change_gcode
M600
; custom gcode end: color_change_gcode

G1 E-1 F4800 ; retract
; acceleration to travel
G1 X148.867 Y148.601 F7800 ; move to first Internal perimeter point (acceleration)
; decel to extrusion
M204 S800 ; adjust acceleration
G1 X150.515 Y150.48 ; move to first Internal perimeter point (deceleration)
; end travel
G1 E1 F4800 ; unretract
;TYPE:Internal perimeter
;WIDTH:0.45
G1 F1511.1324
G1 X150.645 Y150.366 E0.00788 ; Internal perimeter
G1 X150.756 Y150.297 E0.0059 ; Internal perimeter
G1 X150.836 Y150.266 E0.00391 ; Internal perimeter`

Code after I moved the M600 command : 

`G1 X140.997 Y145.188 E0.00402 ; Solid infill
G1 X142.009 Y143.673 E0.08266 ; Solid infill
G1 X144.28 Y141.403 E0.14565 ; Solid infill

;COLOR_CHANGE,T0,#0ED51F
; custom gcode: color_change_gcode
M600
; custom gcode end: color_change_gcode

;LAYER_CHANGE
;Z:1.32
;HEIGHT:0.28
; custom gcode: before_layer_gcode
;BEFORE_LAYER_CHANGE
;1.32
G92 E0
; custom gcode end: before_layer_gcode

;WIPE_START
G1 F6240
G1 X143.361 Y142.322 E-0.95 ; wipe and retract
;WIPE_END
G1 E-0.05 F4800 ; retract
G1 Z1.32 F7800 ; move to next layer (4)
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;1.32
; custom gcode end: layer_gcode

G1 E-1 F4800 ; retract
; acceleration to travel
G1 X148.867 Y148.601 F7800 ; move to first Internal perimeter point (acceleration)
; decel to extrusion
M204 S800 ; adjust acceleration
G1 X150.515 Y150.48 ; move to first Internal perimeter point (deceleration)
; end travel
G1 E1 F4800 ; unretract
;TYPE:Internal perimeter
;WIDTH:0.45
G1 F1511.1324
G1 X150.645 Y150.366 E0.00788 ; Internal perimeter
G1 X150.756 Y150.297 E0.0059 ; Internal perimeter
G1 X150.836 Y150.266 E0.00391 ; Internal perimeter`