supermerill/SuperSlicer #2022: z-offset from plane on build-plate

remopini, 2021-12-08:
### What happened?

It seems that if bottom z-offset is set to something bigger than 0, then this is also used for the first layer of a raft. This makes the raft print above the actual build plate (and thus fail miserably). Tried it with bottom z-offset of 0.2mm.

If I set bottom z-offset to zero, the raft prints fine, but then any supports on the model stick to the model part they are "growing" from.

I believe this to be a bug, as the bottom z-offset should only apply to interfaces between support material and a part of the model, NOT support material and the build plate.

If this IS indeed working as intended, then there should be a way to define a bottom z-offset for supports and model surfaces separate from a bottom z-offset between build plate and supports.

### Project file & How to reproduce

[Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/7677078/Calibration.cube.zip)

### Version

2.3.57

### Operating system

windows 11

### Printer model

artillery x1 with duet/hemera

supermerill, 2021-12-09:
I loaded your porject file and the raft touch the build plate.
[image]

what do you have?

remopini, 2021-12-09:
hm... indeed, I just created two g-code files with the setting changed inbetween and they are identical.. I'll keep trying to figure out where that offset came from...

closed 2021-12-09 by remopini