supermerill/SuperSlicer #2023: Can't de-select built in filament presents

Czarnyszarf, 2021-12-08:
### What happened?

Issue was mentioned to discord - I am unable to hide all filament presets .

When I open the configuration wizard and deselect all filament,and press finish to save the results -  a window pops up wnich says that xxx printer doesn't have filament selected and asks you if you want to select default presets.

You cant close the wizard without selecting at least one filament. 
[image]

### Project file & How to reproduce

Go to configuration wizard (I use ender 3 bl touch profile)
Deselect all the filament presets
Press finish 
Press no on the popup window
Press finish again - popup window shows up again... 

Closing the slicer while having wizard opened selects default 3 presets (pla/petg/abs) 

### Version

Version 2.3.57

### Operating system

windows 10

neophyl, 2021-12-08:
I think the 3 profiles all need at least 1 entry in them.  They cant be null, otherwise that would cause all sorts of software issues. 
 So it needs to install at least 1 filament.

supermerill, 2021-12-08:
but if he has a valid 'user' filament, maybe we can let him select no filament.

Czarnyszarf, 2021-12-08:
yes, I have valid user filament

Why I want to remove system presets - I have set filament presets for profiles 
Like different temperatures for speed profiles, lower temps for quality ones etc.

When I select different printer profiles - it randomly defaults to system preset(generic pla for example).