supermerill/SuperSlicer #2024: Height constraint range for variable layer height

Eliminateur, 2021-12-08:
**Is your feature request related to a problem? Please describe.**
There's no way to configure a +- range of height adjustments for the "variable layer height" function, it defaults to the max/min of the extruder layer height, which is not really desirable

**Describe the solution you'd like**
Add an option to enter the desired limits for +/- variation of the layer height in the "print quality" preset, for example i want X preset to have +-0.04mm variation. Or for another profile i only want a -0/+0.04 for example

**Describe how it would work**
That setting would constrain the variable layer height to the range one desire per profile, the current implementation is too extreme and leads to poor quality prints as for example goes to very high layer heights on straight features that lead to wall gaps as the extruder can't keep up that speed for that height of extrusion as the profile currently selected was not made for it

**Describe alternatives you've considered**
the alternative is to set the range on the extruder everytime i switch profiles

supermerill, 2021-12-08:
> which is not really desirable

why? it's the sole use of these settings, to know what layer heights are printable.

Eliminateur, 2021-12-08:
because those setting are for the extruder, not for the profile, i know my extruder can do from 0.12 to 0.32 for example, that does NOT mean i want the variable layer to go to those ranges on a .28 profile because it will lead to bad prints as the rest of the profile settings are NOT optimized for the extreme layer height difference it currently does with no limit.

In my .28 variable profile i want a +-0.04 variation only as a further example.

But on my 0.12 variable profile i want a -0/+0.04 because i don't want the height to go below what realistically the printer can do but i do want upward movement for speed gain. Having to go switching the extruder settings globally everytime i switch quality profiles is error prone and cumbersome -a workaround at best- to circunvent the lack of limit.

as reference, that's how cura implements the variable height option -per profile-

supermerill, 2021-12-08:
Understood.