supermerill/SuperSlicer #2028: Solid Infill Line Overlap Percentage

blex-max, 2021-12-09:
On certain filaments with strange flow properties, e.g. composite filaments with powdered stone for aesthetic purposes, large sections of solid infill tend to become overextruded even when walls look very good.

**Describe the solution you'd like**
It would be nice to have a setting like perimeter overlap percentage but for solid infill, allowing for the flow rate to be adjusted to suit.

**Describe alternatives you've considered**
A flow ratio percentage for solid infill would also be fine.

supermerill, 2021-12-09:
in filament settings, you have "max overlap". Should do the job.

blex-max, 2021-12-09:
My concern with using max line overlap is that it will also affect perimeters, which come out very nicely with perimeter overlap set to 100% I suppose though since it appears to be a cumulative effect of many lines close together, using max overlap is unlikely to negatively affect perimeters too much.

supermerill, 2021-12-09:
As perimeters are printed first, they aren't as affected visually but they may be pushed outward (and inward a bit) and so are impacted even if it's not visible. They can also contribute a bit to solid infill over-extrusion.
If you can prove me (picture with-without) that this setting is problematic for your perimeters, then we can discuss on how to best solve it.

blex-max, 2021-12-09:
I will investigate! The worst of it is, walls surrounding areas of solid infill on these filaments are often very bumpy as the infill overlap goes too far over the internal perimeter causing the above layer of wall to squish outwards. I didn't mention this in my initial post, I think I should have.

blex-max, 2021-12-14:
So I have been experimenting with this a bit and my conclusion is that it would be really nice to have line overlap percentage options for top, bottom and solid infill in addition to perimeters as is currently implemented. Is this something that could be done? Thanks!

supermerill, 2021-12-15:
> Is this something that could be done?

It can be done. But I'm still wondering if it's really useful. What I propose is that I implement it, share with you a dev build and you can test & report if it's relevant or not?

blex-max, 2021-12-15:
Alright, I would appreciate that greatly!

blex-max, 2021-12-19:
How long do you think this will take? There is no rush, take as long as it needs - I just don't always get notifications from github so I want to know roughly when I should check back!

Thanks for all your work :) Happy Holidays 🎄

mcjohn57, 2022-01-03:
Just wanted to add that I've had this exact issue in printing a lot of PETG.  My solid infills were growing about .05mm per layer.  If I ever printed 4 solid top layers on something really large it would always turn into a mess.  The only compensation that ever worked was the extrusion multiplier.  When dropping below 90% I started to lose a lot of dimensional accuracy, and still had issues with solid infill overflowing.  Not sure if there should be a new "calibration" to adjust this, or if there is a recommended way to calibrate this.  It seems like the current Filament Flow Calibration could be impacted by this same behavior.  Really appreciate the work being done here.

SinkWhole, 2022-01-03:
I also want to say that I have a lot of trouble tuning my material flows.
The filament flow rates which produce a hollow 1 perimeter Cube with the right wallthickness never looked good when trying an flow calibration.
I even went down the route of changing the extruder to a geared dualdriven one, enabling linear advance and tuning it for many conditions and measuring the diameter of the filament I spend on the calibration prints.
I left it at some point and currently use Prusa Slicer because it finally manages to use multiple screens properly.

While reading this I noticed my PolyTerra contains some sort of fillers, which corresponds to the original post.

Right now I try to get the PolyLite PETG running, still without having real succes  on tuning infill and perimeters/accuracy at the same time, corresponding to the last post.

I always fine tuned the different Flow rate settings in Cura to get spot on prints, I think there is a Flow factor on almost any feature type.
Hopefully there is a better final solution on this.
And maybe we just have found a annoying flow effect of unpure polymers.  x)

blex-max, 2022-01-03:
Based on the sponsor poll I suspect that this might not be a priority for a while, but I do see a couple of commits - @supermerill is this feature available in the nightly/dev branch?

blex-max, 2022-01-04:
@mcjohn57 @SinkWhole I've just checked out the nightly build - a new setting called solid infill overlap is present! Without word from Merill I can't say whether it's functional/ready yet but I'm going to give it a try. I believe it will solve my problems, and I suspect yours too.

SinkWhole, 2022-01-07:
@blex-max I just stumbled over infill/perimeters encroachment, which seems to be at 25% at default and says it should not be needed,
as described it sounds like it should only be applied to Sparse infill, since solid infill should be managble by filament amount, not by overpositioning to compensate backlash.

Unfortunately I don´t have a machine to test it out atm, but maybe turning this off could solve the odd overextrusion, if it catches on solid infill for some reason.
In my case, all over extrusion artefacts summed up at the location where the infill will connect to the walls.
If thats the cause we could avoid having flow parameters for every line type. Which is generally nice to have, but a pain of work to tune.
Maybe someone with similar issues could give turning off {infill_overlap} a quick flow calibration?

supermerill, 2022-01-07:
> @mcjohn57 @SinkWhole I've just checked out the nightly build - a new setting called solid infill overlap is present! Without word from Merill I can't say whether it's functional/ready yet but I'm going to give it a try. I believe it will solve my problems, and I suspect yours too.

sorry, forgot to tell you to test it. I had some problems with the nightly, and wanted a less buggy one before tasking you. 
Can you report on your findings?

@SinkWhole 
the `infill/perimeters encroachment,` is the good ol' `infill/perimeters overlap` but as I added many 'overlap' setting that are a totally different kind of overlap, I renamed it so it's now distinct and can't be confused with them.

blex-max, 2022-01-10:
I'll get on to testing this as soon as I can. It might take a few weeks though, I'm not going to have much time for my printers for a bit!

gpread, 2022-01-12:
I have been having this problem with PETG also, solid infill area's being rough when extrusion multiplier is otherwise perfectly set (exact correct width for a single perim wall).  I can get them to clean up by significantly reducing extrusion multiplier (-5 to -10%) but then I have significant dimensional accuracy loss of course.   I didn't seem to have such difficulty (though still significant effort to get settings dialed in) before when I was using PSlicer, so I do wonder if its somewhat a consequence (or compromise) as a result of other useful tweaks in Susi.

So this seems like a useful addition to me (at least single %flow value for all solid area's).

Having multiple settings for different types seems like a good idea, if it does go in that direction then I'd like to see one specific to the first layer (which for me often seems to be one worst affected by this).

blex-max, 2022-01-18:
@gpread try it! it's in the dev build :) I haven't had time to experiment with it yet.

neophyl, 2022-01-18:
Wonder if thats why the default prusa printer profiles always set flow to S95 in their start gcode.

gpread, 2022-01-18:
@neophyl yes, I also wondered that after I found what I did worked nicely for me.

I guess also relevant to say this only becomes a noticeable issue when the solid infill area is quite large (I was working with items that had > 75x75mm area of solid infill. (and I noticed that the infill area would look clean to begin with but then degrade at some point, at worst then blobbing on perimeter at some point before its completed).

@blex-max, I might but have several other things I want to do/fix too and I was kinda wanting to avoid spending any time on another version of Susi until there is a PS 2.4 based version (as I already established there are a lot of things I was/am looking for in that).  Currently I don't have myself setup to make builds (I need Windows 10 x64) of Susi myself, so its non-trivial for me to use dev versions until I do.

blex-max, 2022-02-16:
I am finally back and able to start testing, @supermerill is this available in any builds at present?

supermerill, 2022-02-16:
in the 2.4 beta. You have one in the release page and a more recent in the action/nightly one.

blex-max, 2022-02-21:
Short answer: Yes, it's useful. Thanks so much for implementing!

Long answer: What I'm finding as I experiment with this is that flow, unsurprisingly I think, doesn't vary linearly with line width. I think the reason I needed this feature is that I have large line widths for my perimeters (c.a. 150% of nozzle 0.6mm diameter) compared to my solid infill (c.a. 105%) (I'm using beefy nozzles/heaters/motors of course). In other words, while this new feature for solid overlap solves my problem, it's sort of a bandage on the underlying issue - I think the 'real' solution would be to introduce some kind of scaling factor for flow as a function of line width. There would even a way to calibrate it per this CNC Kitchen [video](https://www.youtube.com/watch?v=UNJdv5bFGOg&t=499s) This has been sort of attempted in Marlin itself with the scaling factor for PID temp as a function of extrusion volume. What do you think?

supermerill, 2022-02-21:
For the mm3/sec -> mflow mult relation, the problem when working on it on the slicer side is that we don't know the real speed (so same for the flow-speed).
Ideally, you have to implement it in the firmware side.
The slicer can help you with calibration prints.

You can sort-of do that in a post-process script where you simulate the firmware so you kind of know at what flow-speed you're printing. But frankly, it's a big hassle as it should be so much easier to implement in the firmware.

blex-max, 2022-02-21:
The speed the toolhead is moving or the speed that the extruder is moving?

Either way, hmm, interesting, so perhaps this is something I should raise on the Marlin repo?

supermerill, 2022-02-21:
The amount of mm3 of plastic extruded per second.

>  this is something I should raise on the Marlin repo?

I think so

blex-max, 2022-02-23:
I think the original substance of this issue has been dealt with, so I'm going to close it - thanks!

closed 2022-02-23 by blex-max