supermerill/SuperSlicer #2029: "extra wipe for external perimeters" not working with only one perimeter

Mandragora-Arato, 2021-12-09:
### What happened?

I am printing a cylinder 15mm in diameter. I use 0,8mm nozzle, 1,2mm external perimeter width, 1 perimeter, 23% infill. SS 2.3.57.6
I wanted to minimise seam so I turned on "extra wipe for external perimeters" by setting 3mm. But on preview nozzle isn't wiping external perimeter at all, it retracts and start to print infill (and correctly doing "wipe while retracting").
When I set perimeters to "2" nozzle is correctly wiping on external perimeter.
Also when "external perimeters first" is ON, wiping on external perimeters does not work even with 2 perimeters being set.
I conclude that there is small bug preventing "wipe on external perimeters" when external perimeters are printed first.

### Project file & How to reproduce

Settings:
"extra wipe for external perimeters" = 3mm
"Perimeters" = 1
Or
"extra wipe for external perimeters" = 3mm
"Perimeters" = 2
"External perimeter first" = Activate

### Version

2.3.57.6

### Operating system

win 10

### Printer model

Prusa MK2S

Mandragora-Arato, 2021-12-09:
Thanks for checking my issue. Is there any time estimate for new version with a fix?

supermerill, 2021-12-09:
you should be able to get the nightly tomorrow morning.
The proper release this week-end

closed 2021-12-22 by supermerill