supermerill/SuperSlicer #2030: Complete Individual Objects

leesafur, 2021-12-09:
### What happened?

New to Super Slicer so forgive me.. I switched from Cura as I couldn't get a project to print correctly with Klipper.
First time ever using it, I set it up for my printer and to my surprise everything worked and printed perfect. Now when I open two files to print individually, the sliced preview shows them being printed at the same time. I finally gave up and sent the Gcode to the printer and it printed them individually.

[image]
[image]

### Project file & How to reproduce

[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/7686166/SuperSlicer_config_bundle.zip)

### Version

2.3.57.6

### Operating system

Win10

### Printer model

Modified CR6-SE BTTSKR6 Klipper

supermerill, 2021-12-09:
How did you failed to create a project file when the whole process is explained in the form for issue creation?

leesafur, 2021-12-09:
Sorry I misunderstood "create a project file" hope this is the right one! 

[18650_8P_base_V2.zip](https://github.com/supermerill/SuperSlicer/files/7686407/18650_8P_base_V2.zip)

supermerill, 2021-12-09:
the sliced preview show you how the slicer slice the objects.
the gcode preview show you want's going to be printed.
How it's ordered is decided when creating the gcode.
That's why the slider behave differently in the two views.

leesafur, 2021-12-09:
LOL feeling pretty stupid right now! Thanks for the tip!

closed 2021-12-09 by leesafur