supermerill/SuperSlicer #2031: "Ironing" Support Interface Pattern Applied To Base Raft (should not be)

scottmudge, 2021-12-09:
### What happened?

When the "Ironing" Support Interface Pattern is activated, the ironing is also applied to the base support raft, rather than just the top support interface layer:

[image]

The fix is probably simple, ensuring the support interface ironing is not applied in the first or second layer.

### Project file & How to reproduce

Add supports for any object, enable supports, use ironing interface pattern, and slice. See attached file
[example.zip](https://github.com/supermerill/SuperSlicer/files/7687659/example.zip)
.

### Version

Latest from "Git Actions" - 2.3.57

### Operating system

Windows 10

### Printer model

MK3S

supermerill, 2021-12-09:
yeah, raft is currently a fake support.
I'll do a pass sometimes after 2.4 merge.

supermerill, 2021-12-09:
also, duplicate of #1512

scottmudge, 2021-12-09:
Gotcha!

Are you planning a full 2.4 merge (including the UI dark mode?) or just partial?

supermerill, 2021-12-10:
full
but maybe for some inner parts that are already updated, but nothing really visible.