supermerill/SuperSlicer #2032: [FR] Align seam throughout all perimeters

kb-elmo, 2021-12-09:
**Is your feature request related to a problem? Please describe.**
Right now the seam alignment is only applied to the outer perimeter. The inner perimeters have a random placed seam.
This can cause fairly long travel moves before the outer perimeter gets printed which can result in "post travel underextrusion" which is especially problematic on bowden style extruders.

**Describe the solution you'd like**
Align the seam throughout all perimeters to the same position.
This way travel moves before the outside perimeter can be avoided completely, resulting in a better looking seam.
It would also reduce the number of unnecessary retractions and travel moves which negatively impact the print time.

**Describe how it would work**
See above.

**Describe alternatives you've considered**
Unfortunately the options for optimizing the seam position and gap are fairly modest in SuperSlicer so there aren't really any alternatives other than using a different slicer (Cura already aligns the seam properly throughout all perimeters and creates far better looking results).

neophyl, 2021-12-09:
Have you actually tried the seam modifier ?  It’s something only present in SuSi.

supermerill, 2021-12-09:
discussion about it: https://github.com/supermerill/SuperSlicer/discussions/1816

kb-elmo, 2021-12-10:
> Have you actually tried the seam modifier ? It’s something only present in SuSi.

Thank you. That actually worked.
I didn't know about this feature.

Yeastdonkey, 2022-11-07:
I would also like this feature. Here's an example of the additional travel move between the inner and outer seam positions when using rear seams. It also causes a slight surface blemish on the outer perimeter from the inner seam.
[image]

The seam position modifier nor the paint-on z-seam aligned the inner/outer seams, although a seam modifier does (second picture). It would be nice to have an option for it though.
[image]