supermerill/SuperSlicer #2037: Stiched Seams

FDResearch, 2021-12-10:
Pretty simple to explain feature request, when the seam position to set to aligned, rear, or cost based to have an option to offset the position alternating each layer by a few mm back and forth to prevent a weak point at the seam. Would be nice to be able to change the distance as well.

supermerill, 2021-12-10:
> to change the distance as well.

what does it means?

FDResearch, 2021-12-10:
May be easier to only make it work for some options since some of those do get a bit complicated. Since I use rear mostly (which also a fan of the front, left, right etc but I see that is already another feature request) I had envisioned the rear just places it alternating back and forth on either side of an invisible rear centerline (where the seam would be currently) so with a small min offset like 6mm you have overlap but the seam is still mostly at the rear where it can be hidden. Maybe just setting the min distance to a very high number or -1 would just put the seam as far as possible from the one below it (although this is probably only useful for very specific applications). Small distances centered around where the seam would have been to hide it without giving up strength is probably the most useful case?