supermerill/SuperSlicer #2038: Gap Fill Issues

CCS86, 2021-12-10:
### What happened?

Gap fill is being added in areas I do not expect, and is generated in a way that causes over-extrusion:

[image]
[image]
[image]

Better here, but weird:
[image]

### Project file & How to reproduce

[Gap-fill-issue.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7695762/Gap-fill-issue.3mf.txt)

### Version

2.3.57.6

### Operating system

Windows 10 x64

supermerill, 2021-12-10:
select a not-`(filled)` for the solid pattern. (use expert mode to show it)

CCS86, 2021-12-10:
Is this only caused with a sparse infill and filled solid pattern?

closed 2021-12-22 by supermerill