supermerill/SuperSlicer #2039: Only show Print Settings to selected Filament Type

Tahx, 2021-12-11:
**Is your feature request related to a problem? Please describe.**
Each filament spools different print settings on some specific parameters, therefore I need to save an entire Print Settings just for that spool. This is cluttering the dowpdown list and it is super annoying to find out you selected the wrong print settings after the print has started.

**Describe the solution you'd like**
There is an entire library of pre-set materials inside the parameter filament_type. I'd like an optional parameter inside the Print Settings->Dependencies that allows me to only show this Print Settings for the specified filament_type parameter.
Please note the current materials library has "PET" instead of the most commonly printed "PETG", is also lacking TPU and all other CF-based filament. I'd only suggest to add "PETG" and "TPU" to the library to make it at least complete for the most commonly printed filament materials.

**Describe how it would work**
Example: I got one filament profile named "ABS - Vendor_Name - Black" and filament_type set to ABS.
In the print settings I wanted to be only shown for this profile, I'd go into Print Settings->Dependencies, tick the box "Filament type dependencies" and select which material in the dropdown list. In this case I'd select ABS.

**Describe alternatives you've considered**
There is no such features as far as I know.

**Additional context**
Thanks.

supermerill, 2021-12-11:
currently, the opposiite is true: a filament can be shown only for compatible print profiles.

Tahx, 2021-12-11:
> 

Thanks for your reply.

I would say it just makes way less sense.
Once a filament is calibrated, it doesn't need modications anymore, so there's no point in having many different filament profiles for each compatible print profiles.

supermerill, 2021-12-11:
I think the rationale is:
I make a printer
then I make print profiles for it.
Then I make filaments with temp & cooling tuned for the print profile.

Tahx, 2021-12-12:
> 

I disagree with that rational.

The logic says that before you start printing, you'd select a filament first, be it whether PETG, ABS, PLA... then you'd select what print settings you go with tuned for that filament.

The tuning of the filament itself is a separate previous step, it has its own properties linked to his material. Once tuned, it doesn't matter anymore. No one is going to change the extrusion multiplier after the spool has been properly tuned.

Circular dependencies is a bad idea, I'd suggest removing the old way in favor of the suggested one. I'd be really curious of who is actually using the current way and for which reasons. At worst they'd probably be happier with the new way.

The suggested workaround is a PITA, the ergonomic isn't good because it is all about the print profile dropdown.

Since the mechanics is already implemented, it shouldn't be too much to change it, if you ever decide to approve the change.

At the end, it's "only" an ergonomic improvement, but to me it's more logical to select the spool first rather than the print settings first to avoid having a clutered dropdown list for the print settings.

To be clear here is an example:
```
regular Speed 0.1 ABS
regular Speed 0.2 ABS
regular Speed 0.3 ABS
High Speed 0.1 ABS
High Speed 0.2 ABS
High Speed 0.3 ABS
Another nozzle width 1 ABS
Another nozzle width 2 ABS
Another nozzle width 3 ABS
Another nozzle width 4 ABS
Another nozzle width 5 ABS

regular Speed 0.1 PETG
regular Speed 0.2 PETG
regular Speed 0.3 PETG
High Speed 0.1 PETG
High Speed 0.2 PETG
High Speed 0.3 PETG
Another nozzle width 1 PETG
Another nozzle width 2 PETG
Another nozzle width 3 PETG
Another nozzle width 4 PETG
Another nozzle width 5 PETG
```

That's a lot of print settings starting to stack up, right? And I hate having to add the name of the material in the name.

Usually they are not sorted in this order, but rather something like ABS, PETG, PLA, PLA, ABS, PETG and so on, they're all mixed up together.

Well, I used to spend a lot of time on the UI and ergonomic back in the day, now I only make command line apps.
I know how quickly it becomes bothersome to search and click on the wrong thing, too often.
Most of SS has good ergonomics but this profile management... can be improved.

supermerill, 2021-12-12:
I can't remove the old way without breaking PrusaSlicer compatibility, so I won't do that.

Tahx, 2021-12-12:
> I can't remove the old way without breaking PrusaSlicer compatibility, so I won't do that.

Oh ok, then how about making it optional?
When one is selected, the other can not, the checkbox is disabled that's it.

supermerill, 2021-12-13:
The code that manage dependancies is already a bit complicated as when something change, it trigger and check visibility of everything and then try to get the best one for each type of profiles.
It's not something I want to touch again. Debugging an issue on it was painful enough.

Adding into that a conditional that erase a bit of the behaviour and dependency chain to something else... I don't want to do it.

I understand the issue.
if someone succeeded to alleviate it without side-effect, I'll merge it.
or I may in the future, when I'll modify again something in this area, maybe I can manage to add this kind of 'dependency'.

Tahx, 2021-12-13:
> The code that manage dependancies is already a bit complicated as when something change, it trigger and check visibility of everything and then try to get the best one for each type of profiles. [...]

I completely understand what you mean by the pain haha. There's a few checks to do, I put it down on text first to have a proper picture of it so I don't confuse myself especially on large UI.

Wish I could do it, I also have some better ideas to improve that part of the profile management, but I don't know a sh*t about cpp despite having done some C for many years, now I mostly do C# and golang, I can't get the grasp of cpp.

Crossing fingers towards the future.