supermerill/SuperSlicer #2042: SuperSlicer does not compile with current versions of TBB

acwest, 2021-12-11:
### What happened?

TBB as of version 2021.1 has been heavily refactored, and the SuperSlicer codebase no longer compiles. The first issue is that FindTBB.cmake no longer works, as it uses tbb/tbb_stddef.h to determine version of TBB that is installed, and that information is now located in oneapi/tbb/version.h. I was able to get that working with some modifications to FindTBB.cmake, but compilation then fails as a number of deprecated functions have now been removed: [https://www.intel.com/content/dam/develop/external/us/en/documents/tbbrevamp.pdf](https://www.intel.com/content/dam/develop/external/us/en/documents/tbbrevamp.pdf)

### Project file & How to reproduce

- install current version of tbb (2021.1 or higher).
- build SuperSlicer

### Version

2.3.57.6, https://github.com/supermerill/SuperSlicer/commit/616fc2b521c2647de7fdfba0de03fdeb5ce2ac7d

### Operating system

Arch Linux

supermerill, 2021-12-11:
I'll update when prusaslicer will update. Or this will make merging across ou branch even more difficult.
Maybe they already updated it on 2.4

acwest, 2021-12-11:
I looked, they don't seem to have updated yet. I can file an issue with them, though

Lukas1818, 2021-12-12:
If some one have this problem, I have apply the [arch prusa slicer tbb path](https://github.com/archlinux/svntogit-community/blob/packages/prusa-slicer/trunk/prusa-slicer-tbb-2021.patch) to Superslicer:
https://github.com/Lukas1818/SuperSlicer/tree/tbb21
You can use it as [patch](https://github.com/Lukas1818/SuperSlicer/commit/75320b9620b5240cda2c9d92cde394fb68ff97b6.patch).

I have also update the aur packages. So they should compile again, without modification.

Lukas1818, 2022-02-05:
this issue seems to be fixed with v2.3.75.10. Superslicer dose build again without any patches.

supermerill, 2022-02-05:
thanks to @RealDeuce