supermerill/SuperSlicer #2043: Modified (shrinking) vertical holes need to be cooling

vladi552, 2021-12-11:
When the internal modified (shrinking) vertical holes with small diameters is prepaired, we need to force to cooling them once the internal perimeters is extruded. Otherwise, the filament haven't enoth time to be harded and the internal hole diameter will be smaller even it has been shrinked.

supermerill, 2021-12-11:
Sorry, i don't understand. Can you explain me with some quick drawings?

vladi552, 2021-12-11:
Sorry for my english.
The SuperSlicer have an awesome option to shrink the holes:
[image]
I have an idea to not only shrink the holes, but also cool them (fan 100% PWM) while the internal hole perimeters is printed:
[image]
Currently, only hole shrinking is implemented.

vladi552, 2021-12-11:
The holes 2-7mm is very sensitive related to this problem.

vladi552, 2021-12-11:
Also I noticed, that the current version of SuperSlicer can generate gcode files where the cooling fan do not switch off at the end of printing job. It a bit difficult to recreate.

supermerill, 2021-12-12:
> cool them (fan 100% PWM) while the internal hole perimeters is printed:

kinda like the "External perimeter fan speed", but only for the "hole" one?
And I'm sceptical about the inner (yellow) one, as in your picture it's simple but often, even the first one can be split and run along the whole object. Or I have to add a threshold.

> Also I noticed, that the current version of SuperSlicer can generate gcode files where the cooling fan do not switch off at the end of printing job. It is a bit difficult to recreate now.

add M107 to your custom end-gcode, just in case.

vladi552, 2021-12-12:
> kinda like the "External perimeter fan speed", but only for the "hole" one?
External perimeter fan speed, but only for the **INNER** "hole" one.

supermerill, 2021-12-12:
?