supermerill/SuperSlicer #2045: Please add support for diff between configuration

zapta, 2021-12-12:
**Is your feature request related to a problem? Please describe.**
It's difficult to understand the differences between saved configurations. Even a diff between current settings and last saved configuration is tricky because when I click on the X button to close the window, if there are not changes, it is actually closed, loosing the models placement.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A full support for diff between configurations. I think the underlying logic already exists so it's just a matter of wrapping it with user friendly UI.  The diff is between an combination of a pair of made of  the current settings, last saved configuration, any selected snapshot.

**Describe how it would work**
Be as descriptive as your skill level allows you.

Add a menu option 'Configuration diff' that will open a window that allows to select the two configurations to compare and shows the diff between them.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I am currently use the close button X but it's limited (can't compare snapshots) and risky (exists if there are no changes).

**Additional context**
Add any other context or paint drawing about the feature request here.
[image]

neophyl, 2021-12-12:
I think that has already been added to PS 2.4.  As such when that is released and SuSi is merged afterwards then I would expect it to also become available after that.

zapta, 2021-12-12:
Thanks. I am using 2.3.56. Will look for it.

zapta, 2021-12-12:
Is this PS 2.4 new feature mentioned here
https://github.com/prusa3d/PrusaSlicer/releases ?  I can't find it.

neophyl, 2021-12-12:
I think you missed my point. Super Slicer usually gets merged after Prusa Slicer versions are fully released. The profile compare was introduced into Prusa Slicer 2.4, which has not officially been released yet. Once it is, at that point Merrill usually starts the process of merging it into the Super Slicer code. 
This can take awhile , especially if there have been many changes, which there have been for PS2.4. 
So it won’t be in there yet but it should get added in the future.

zapta, 2021-12-12:
I sent you the release notes of PS 2.4 which is already in advanced beta. I
couldn't find in the PS 2.4 release notes any mention of the diff feature.

Are you sure that it will be included in PS 2.4?

neophyl, 2021-12-12:
https://github.com/prusa3d/PrusaSlicer/issues/6458  Was added into master back in April.

zapta, 2021-12-12:
> prusa3d#6458 <https://github.com/prusa3d/PrusaSlicer/issues/6458> Was

Thanks.

neophyl, 2021-12-13:
Its been in PS since the alpha releases (still had alpha 2 on my HDD).  Under Window>Compare Presets.  Still in the Beta 4 too.

zapta, 2021-12-13:
Does the prusa implementation allow comparison between configurations (e.g.
snapshots) or just between presets of the currently loaded configuration
(which is not that useful from version management perspective) ?

neophyl, 2021-12-13:
It allows you to select configurations in the drop downs, so you can select 2 different print presets, 2 different filament presets and 2 different printer presets. Then you can click the compare button.   I don’t think it works with the current selected profile, unless you save it as it compares the saved ones. 

Feel free to download the PS beta.  As you don’t need to install and it uses its own beta config folder it won’t effect anything on your system. You can see if their implementation does what you want.

kevenwyld, 2022-08-23:
I just went looking for this feature and found this issue.  This diff functionality is now available in the latest release.

The only improvement I can think of would be to allow generating a diff from current settings rather than having to save them to compare.  Similar to what is done as part of the close dialogue when there are unsaved changes in a preset.  That might be a ton of work for little gain though since you can just quit without saving and see the same thing. 

Here's what compare looks like in the latest release, you can also click the compare button next to the search button at the top of the settings tab.:
[image]