supermerill/SuperSlicer #2047: Seam in vase mode

spartanJon089, 2021-12-13:
### What happened?

Vase mode creating weird artifacts or "seams". appears as gaps or misaligned layers in the gcode previews. 

[image]
[image]
[image]

### Project file & How to reproduce

[VaseMode.zip](https://github.com/supermerill/SuperSlicer/files/7700835/VaseMode.zip)

### Version

Version 2.3.57.6

### Operating system

windows 10

### Printer model

modified ender 5 +

supermerill, 2022-01-31:
currently, the vase mode is crude:
it continue to slice layer by layer
then it will change the z based on the distance % on a layer. So when it's at the top (end of the layer, it is at the same height as the next one, but not at the right x & y if there is a slope.
I tried to move the seam away from slopes, but there is no start position that won't push you into a slope here. So it's an unavoidable artifact of the current slicing algorithm.

robustini, 2022-03-30:
@supermerill please check that project with the nightly, i can't slice it, empty result.

supermerill, 2022-03-30:
@robustini 
if you put `G1 Z0.01` after the START_PRINT in the custom start gcode AND you uncheck the `use_relative_e_distances`, then the gcode viewer works again.

I don't why yet, I'll look into it.

robustini, 2022-03-31:
> G1 Z0.01

Thanks for the tip, that string after the start gcode was enough for me, the other option that is flagged or not in my slicing does not make a difference.

BlackNet, 2023-05-10:
I have noted this bug as well.  When I went digging for what / why I found something interesting.  This is a known bug in PS for several years now.   https://github.com/prusa3d/PrusaSlicer/issues/2841  This is from 2019, there is a 'fix' listed but it's not a good fix at all and it does mess up the model.

Dougwar, 2024-04-04:
any news about it?