supermerill/SuperSlicer #2051: Height Range Modifiers to select Different Print Profiles

blex-max, 2021-12-13:
**Is your feature request related to a problem? Please describe.**
Since PrusaSlicer (and by extension SS, though it is not the fault of SS) tends to do odd things with solid infill and top/bottom layers I use height range modifiers a lot when printing to add solid sections etc. This often means changing a lot of settings in a given height range, which can be very laborious

**Describe the solution you'd like**
I'd like to be able to select different print profiles within height range modifiers, if it's possible.

As always, thanks for your work!

neophyl, 2021-12-13:
Are you aware of the ‘easy’ modifier workaround ?
One of the things it’s possible to do is to copy/paste modifiers from one object to another using the right hand tree. Select you modifier, control+c and then select the target object and use control+v.  All the modifiers can be copied across in this way. 

One of the things some of the users on the Prusa forums do is create some very small objects with their common used modifier combinations, set them as non printable and then place them off the plater. They then save the project. 

The workflow then is to load that saved project instead of using new project. They then can load objects, use the copy /paste modifier settings to quickly apply them to the new objects and off they go. 
It’s a bit of a hack but it works and is a very useful work around.

blex-max, 2021-12-14:
Thanks for this, I wasn't aware of it. But as you say, it's a very hacky fix - is there no chance of implementing the feature I'm suggesting?

supermerill, 2021-12-15:
It may a bit deceptive as not all print profile settings can be applied to a height range modifier (even less than for an object).
What may be possible without too much hassle:
 * I can add an entry "Add modifiable settings from" and then you select the print profile name
 * it will add every modifier-compatible settings that are different from the current print profile inside the modifier.

blex-max, 2021-12-15:
That would be fantastic and I would greatly appreciate it!