supermerill/SuperSlicer #2055: Mac Builds are broken

spuder, 2021-12-13:
### What happened?

The current builds are corrupted for OSX

```
shasum -a 256 SuperSlicer_2.3.57.7_macos-arm_211212.dmg
be332f90c83c26c78a5fed8dd4e8a79b479ca92726874b6f4889bb69ae6f2a48 
```

### Project file & How to reproduce

Download latest build from here: 

https://github.com/supermerill/SuperSlicer/releases/tag/2.3.57.7

Try and run on an M1 mac

By comparison the Intel builds run fine.

I've tried downloading several times to verify it isn't a corrupt download

[image]

### Version

2.3.57.7

### Operating system

OSX

spuder, 2021-12-13:
Interesting. I don't have to run that command for intel builds. 

I tried `xattr -cr /Volumes/SuperSlicer/SuperSlicer.app`, but its still giving the same error

samwiseg0, 2021-12-13:
from the screenshot you did not install it into the applications folder. Make sure you install it/move it there first. Do not run it from the disk image

spuder, 2021-12-13:
🤔  So that worked, but I think there is some improvement that can be done here since that shouldn't be required to run. 

1. I've never had to run `xattr` for any other app on my m1 mac. That seems like an unnecessary hoop to jump through. 
2. Intel builds don't require this manual change. 

I'll also bring up the question, why are there separate builds for arm at all? I do appreciate all the wok that has been done to create ARM builds. I think separate builds are unnecessary because its possible to create universal builds (which is currently what prusa slicer does)
[image]

samwiseg0, 2021-12-13:
> 🤔 So that worked, but I think there is some improvement that can be done here since that shouldn't be required to run.
>
> 1. I've never had to run `xattr` for any other app on my m1 mac. That seems [...]

The improvement would be to properly sign the application. I doubt that would happen in this case. 

That is because they sign their executables. I doubt the developer would want to jump through that expensive hoop.

supermerill, 2021-12-14:
I'm trying to get an inexpensive signing certificates.
But I fear it won't work for macos anyway, as I think they want me to own a mac to sign the app.

kgleason, 2021-12-17:
AFAIK Apple won't let you work with any signing certificate off the shelf. You have to sign up as an Apple developer, and use their signing cert. It's been a minute since I've done any of it, but as I recall it was like $100 / year to sign up to be an Apple dev. Back in the day, you absolutely had to use XCode to sign things. I think that has relaxed of late, but I don't know that you can sign an Apple app on Windows.

The risk with universal builds, especially early on, is that you can't know if your universal build will break across all platforms. If you have a stable code base (intel), and you want to try a platform (ARM), I think it makes sense to do separate builds until all of the kinks are worked out of the process of the new platform.

I'll do some testing on my M1 this weekend to see how it plays.

ibash, 2022-01-03:
Github actions have a vm for macOS, so it should be possible to let github handle signing: https://docs.github.com/en/actions/deployment/deploying-xcode-applications/installing-an-apple-certificate-on-macos-runners-for-xcode-development

supermerill, 2022-01-04:
but that means I have to give the signing key to GitHub, so it's public... so the purpose is defeated.

Trackhe, 2022-04-06:
I had the Problem too. xattr -cr the app works for me aswell

michaelmyc, 2022-05-04:
We might want to set up a donation for this purpose and highlight it in README so @supermerill won't need to pay the $100 Apple tax out of his own pocket. At the very least, we should highlight the `xattr -cr /Applications/SuperSlicer.app` workaround in the README.

supermerill, 2022-05-04:
> We might want to set up a donation for this purpose and highlight it in README so @supermerill won't need to pay the $100 Apple tax out of his own pocket. 

Already done, you can donate with the "sponsor" button here.
I paid the apple tax. Now I'm trying to figure out how to properly sign the app. My problem is that I have only access to a mac computer one time each week.

michaelmyc, 2022-05-04:
> My problem is that I have only access to a mac computer one time each week.
I recently got a M1 Mac that I'm planning on doing all my personal projects (including 3d printing stuff) on. Happy to assist if you need to run mac-specific stuff. Also happy to help with M1 testing and other stuff Mac-related.

supermerill, 2022-05-04:
I have to understand the signing stuff myself, i can't share my id.

supermerill, 2022-05-15:
Here the app & dmg for arm mac, signed.
https://mega.nz/file/YxIyXaxJ#34l9oZLe__2tnLy0QfEfm9ZKdkMDgQYz4T8XtBql2pU
With these, you shouldn't need any xattr things.

issic, 2022-05-16:
Awesome! I as well have several M1 and M1ProMax computers for testing if you need.

th3fallen, 2022-07-14:
@supermerill this is still occurring in the latest release 2.4.58.3

supermerill, 2022-07-15:
@th3fallen The 2.4.58.3 mac arm isn't signed?

jfleschler, 2022-07-15:
@supermerill I'm seeing the issue with the latest released version as well.

supermerill, 2022-07-15:
Can you try with this one?
https://mega.nz/file/dogUmRBL#W3t0ZdDiq9YgbErf2L_bDII_cyeSoYo6Mudc3jNzsqo
I verified, and it's signed.
Maybe I have to fresh my Mac certificate...

supermerill, 2022-07-15:
weird.
It's the same as the one i put in the mega archive. Just renamed.
Maybe github modify something... I'll try some things.

alexvonsiebenthal, 2022-07-15:
The file from Mega doesn't work either. There are 2 apps in there: one directly in the zip, that won't open because it can't verify the developer:

[image]

One in another DMG, that can't open for the other error:
[image]

supermerill, 2022-07-15:
Ok, I finally understood what was wrong, and learned how to do it completely.
The dmg from the release page should work, can you test it?

alexvonsiebenthal, 2022-07-15:
> Ok, I finally understood what was wrong, and learned how to do it completely. The dmg from the release page should work, can you test it?

Yes it works!

> For the universal app, I don't know for now. The intel one is made automatically by a GitHub script, and I have to create the arm manually on an arm Mac. I don't succeed anymore to cross-compile. Last months, I spend the majority of my time messing with Mac and apple instead of doing anything on the slicer... won't do the universal app myself in the current year.

That's absolutely fair and just a minor thing that can wait IMHO. I just raised the idea because Prusa is doing it and I have no idea how much of a hassle it is to merge this process into SuSi.

supermerill, 2022-07-26:
Hi
here is a test for an universal app:
https://mega.nz/file/01BAVLyJ#D36IjInHUg5_l-uW-x4XY7BRhsQS4jLioeDR5517Shs
can someone:
 * on INTEL: run it without the "brew zstd" (uninstall it if installed), as it should't be needed anymore
 * on ARM: run it after a "xattr -cr /Applications/SuperSlicer.app" (it isn't well signed yet)
 and report.
 Thanks

ZerrafRX, 2022-08-06:
Universal for ARM is not working with the same error

closed 2024-02-11 by supermerill