supermerill/SuperSlicer #2057: Printhead crashing on Bridge Flow Calibration

AlexHuebi, 2021-12-14:
### Version
2.3.57.7

### Operating system type + version
Windows 10 21H2

### Behavior
The Generated Files changes the settings to print one preset after another, this leads to the printhead crashing into the first printed element on printers with larger printheads or ones that don't have the necessary clearance. 

To solve this, the generator should not change the settings and print everything at the same time or stack the parts like in the temperature calibration.
Another solution would be to include some settings to set the printhead size and get the slicer to pay attention to the printhead limitations.

supermerill, 2021-12-15:
staking them is a good idea.

AlexHuebi, 2021-12-22:
I created my profile from one of the default Bowden presets.
[image]

I also made a sketch to help visualise the needed clearances for my printer
[image]