supermerill/SuperSlicer #2058: No Thumbnails generated in CMD-Line Version of Superslicer

Rincewind76, 2021-12-14:
### Version
2.3.57.7
(but also visible in older versions)

### Operating system type + version
Windows 11

### Behavior
Problem description:
Using the CMD-Line Version of Superslicer produces identical GCODE compared to GUI version, but does not create a thumbnail.
(Necessary files in attached ZIP File)

Reproduction:
Execute the following command in the command line:
"C:\Program Files\SuperSlicer\superslicer_console.exe" -g 3DBenchy.zip.amf --load voron_04_ABS.ini --output 3DBenchy_0.4n_ABS_with_CMD_Line.gcode
The output file does not contain a generated thumbnail

If I use the GUI:
Open Superslicer
Import Config "voron_04_ABS.ini"
Slice Project
Export GCODE
The output file contains a thumbnail as specifided in INI file, rest of the GCODE is identical to CMD-Line version

#### Project File (.3MF) whe
[Superslicer_CMD_Line Thumbnail.zip](https://github.com/supermerill/SuperSlicer/files/7711341/Superslicer_CMD_Line.Thumbnail.zip)

neophyl, 2021-12-14:
Please see https://github.com/supermerill/SuperSlicer/discussions/1994

supermerill, 2021-12-15:
tl/dr: can't take a picture for the thumbnail if there is no GUI to render it.