supermerill/SuperSlicer #2059: separated infill

khunpoum, 2021-12-14:
### What happened?

infill divided to segments
[image]

### Project file & How to reproduce

need to unite infill

### Version

2.3.57.6

neophyl, 2021-12-14:
Without an attached zipped up project file as requested in the instructions when opening new issues anything will be a guess.  Might be a good or a bad guess, who knows.

The type of infill is usually determined by the layers above but some settings can influence this.  Is there geo above those areas ?

neophyl, 2021-12-15:
Thanks for attaching a project.  I'm going to sound a bit pissy here :) but the one you attached is not the same one you pictured.  As such when I opened it and sliced in the same software version you used I did not get the disjointed solid infill (the purple infill).  That was unbroken around the hinge area.  That is not very helpful when trying to chase down issues.

The sections with internal infill (dark red in colour) were separated though.  This is normal and expected as there is geometry above those areas.  The profile specifies 3 top layers, as such it changes the infill type at the lower layers to give 3 top layers for those features.  So it goes through and does any supporting layers for the needed layers above to fulfil the settings.   As all the extra squares in a grid are higher, then those start the transition at a higher layer leading to the 2 different types of infill.  That's why I asked about geo.  While in this example the extrusion is the same the 2 types are defined differently and could in theory have different extrusions widths or other properties.  Of course our human brains go, whats the point those are the same.

I was more concerned with the split solid infill you had pictured, but as I say the project you posted doesn't have that so that's still an unknown.

As I also mentioned though there are some things you can do to control how a slice is done by altering other settings.  You could use a modifier etc to adjust things but in this case simply adjusting the amount that solid infill is anchored can be used to make that surface unbroken.  Setting Print Settings>Anchor Solid infill by X mm to 10mm will do the job.  I've attached a modified version of your project to illustrate

[infill-bug-at-layers-14-16_neo.zip](https://github.com/supermerill/SuperSlicer/files/7717593/infill-bug-at-layers-14-16_neo.zip)

supermerill, 2021-12-15:
It's probably the over-bridge flow ratio splitting area where it needs a bit more flow, as this can't be integrated in another way right now.

khunpoum, 2021-12-19:
another example is attached
[image]
[Bottom_v2_rear.zip](https://github.com/supermerill/SuperSlicer/files/7741703/Bottom_v2_rear.zip)

supermerill, 2021-12-19:
It's 100% the over-bridge flow ratio splitting area where it needs a bit more flow, as this can't be integrated in another way right now. Set it to 100% if you don't like these.