supermerill/SuperSlicer #2060: ERROR: First layer height can't be higher than min layer height

sleeperninja, 2021-12-14:
### Version
_SuperSlicer version 2.3.57_

### Operating system type + version
_Windows 10 (21H1)_

### Behavior
* Description: _When choosing 0.05mm ULTRADETAIL print preset for at least the Original Prusa MINI & MINI+ printer preset, the **Slice now** button changes to **Invalid data**, and the error message in the title shows up in the bed view. Did not test with other printer profiles. Occurs regardless of filament present chosen._
* Steps to recreate: _Open app. Add 3DBenchy to print area. Select Original Prusa MINI & MINI+ printer preset, choose any filament preset, then choose 0.05mm ULTRADETAIL print preset._
* Expected outcome: _Ability to slice._
* Actual outcome: _**Slice now** button becomes disabled, button text changed to **Invalid data**. Error from title on print area view._

#### Project File (.3MF) where problem occurs
Used 3DBenchy.stl
[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/7712727/3DBenchy.zip)

[image]

sleeperninja, 2021-12-14:
Just for clarification, I was doing a side-by-side comparison of file slices between SuperSlicer and PrusaSlicer, and even though both printer presets list 0.07 as the min layer height, PrusaSlicer doesn't seem to care if you select the 0.05mm preset.

neophyl, 2021-12-14:
The error message is misleading anyway, as its not the first layer height that is causing the problem (as that is set to 0.2 anyway).  Its the normal print layer height.  
The way I see it is that its PS that is in the wrong here.  The profile has a 0.07 minimum layer height as defined in the extruder but its being told to print layers that are 0.05 high.  That SHOULD generate an error, if PS isnt doing so then that needs to be pointed out in the PS github.

supermerill, 2021-12-15:
bad ctrl-c ctrl-v. 
error message fixed.

scotte, 2022-01-17:
Just adding a note in case anyone else runs into this the way I did. I use a configuration based on Dr. Lex profile for the FFCP[1] which has a `First layer height` that is larger than the `Base layer height`. Reducing the first layer value to the base layer value resolved it for me.

[1] https://github.com/DrLex0/FFCP-GCodeSnippets/tree/master/ConfigBundles

closed 2022-02-13 by supermerill