supermerill/SuperSlicer #2064: Skirt and supports create insane E-values

srbjessen, 2021-12-15:
### Version
2.3.57.6

### Operating system type + version
Windows 11 - 21H2

### Behavior
When setting skirt active while supports are enforced, The resulting e-values are insane, to best describe it.
* _Steps needed to reproduce the problem_
Activate supports and skirt in the orientation as in the attached file.
* _Expected Results_

* _Actual Results_
[image]
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
[Error skirt.zip](https://github.com/supermerill/SuperSlicer/files/7719546/Error.skirt.zip)

neophyl, 2021-12-15:
When I slice your project in 2.3.57.6 on windows 10 I get quite sane results.

For example
M117 Time Left 5h57m22s
M201 X1500 Y1500 Z120 E10000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z10 E120 ; sets maximum feedrates, mm/sec
M204 P250 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X2.00 Y2.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M140 S60 ; set bed temperature
M104 S235 T0 ; set temperature
M104 S220 T1 ; set temperature
;TYPE:Custom
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
T0
G92 E0
; Filament gcode
M104 S235 T0 ; set temperature
M104 S220 T1 ; set temperature
M109 S235 T0 ; set temperature and wait for it to be reached
M190 S60 ; set bed temperature and wait for it to be reached
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 Z0.2 F9000
G1 E-2 F2400
G1 Z3.2 F9000
G1 X120.318 Y139.909
G1 Z0.2
G1 E0 F2400
;TYPE:Skirt
;WIDTH:0.4
G1 X120.376 Y139.87 E0.00192
G1 X120.507 Y139.816 E0.00578
G1 X120.576 Y139.802 E0.0077
G1 X121.579 Y139.802 E0.03512
G1 X121.648 Y139.816 E0.03704
G1 X121.792 Y139.878 E0.04131
G1 X121.935 Y139.816 E0.04559
G1 X122.004 Y139.802 E0.04751
G1 X123.008 Y139.802 E0.07492
G1 X123.077 Y139.816 E0.07684

So I don't know why I'm not seeing the same.  If you load in your saved project does it still do it ?  All I can think of is that theres some difference between the profile saved on your pc and that exported to the project.  Theres been a few cases in the past with Prusa Slicer where the user has had a malformed/corrupted profile folder and its sliced with those different settings on their machine but their project has been fine when loaded by others.  

You could start Super Slicer using the datadir switch and set up a local set of profiles so its 'clean' and then load your saved project in and check what its doing. 

I notice that you are using a lot of layer heights and extrusion widths based on percentages rather than fixed values in your profiles.  All it would take is one base number loaded/saved wrong for all those % to be based on and you suddenly get extrusion widths and layer heights that are not what you are expecting.

srbjessen, 2021-12-15:
It makes sense with the %.
I had a feeling something might be screwed with my profiles, just no idea how to test it.

Will try your suggestion. You can close the issue as it's obviously only mine that causes issues :-)
Thanks for testing

supermerill, 2021-12-15:
> You can close the issue as it's obviously only mine that causes issues :-)
Thanks for testing

If it's reproductible, it has to be fixed.

srbjessen, 2021-12-16:
Reloading the file still results in the same high values. Will try without percentages first to rule out one thing at a time, i will let you know what i find.

srbjessen, 2021-12-16:
After a bit more testing:
All the values are set to % again, except for "first layer". When setting this to a percentage, even the default 140% it immediately ruins the E and time values to the same crazy high numbers. Setting it to anything in a normal decimal way works perfectly. I will try to build a new profile, and see if it's just a "missing comma" og something else somewhere in the profile-codes. Just noticed a new version is out, will try to update and see if it fixes it too.

supermerill, 2021-12-16:
> see if it's just a "missing comma" og something else somewhere in the profile-codes.

even if it's 'only' that, that's something i have to detect to prevent problems.

I tried to replicate but i failed.
If you save your project file without the first layer "percentage", then laod it and modify to &40%, you replicate the bug consistently? if yes, can you share this file?

srbjessen, 2022-01-11:
I've been unable to reproduce problem with version 57.8. So i will close issue, as it's probably some file that got corrupted

closed 2022-01-11 by srbjessen