supermerill/SuperSlicer #2068: FR: Calibration for G1 per Second

CCS86, 2021-12-15:
With the new and very useful "Maximum G1 per Second" machine setting, we have great control of what is being fed to the motion controller.

The biggest problem most people will find is coming up with the best number to put in there. The true maximum rate can be obscured during normal prints, because longer line segments let a depleting buffer catch up. I think the "default" value of 1500 is wayyy too high for most printers. On my Prusa i3 MK3, I run it at 100. and on my Duet powered machine 350.

I think a calibration for this would be perfect. You could just print a cylinder (or another model with a sequence of different radii) in vase mode, at a print speed you like, and tell the calibration a starting G1/sec and ending G1/sec. It could change the setting to increase layer by layer.

The end result would be so useful, where you can see the optimal setting for your machine. The print would start out with visible faceting, then as it builds up the facets fade away and eventually you choke the controller and see zits.

supermerill, 2021-12-15:
Good idea. Will do it.

>  think the "default" value of 1500 is wayyy too high for most printers

Of course, but there is firmware that may handle it, like klipper. I didn't want to hinder them.

> On my Prusa i3 MK3, I run it at 100. and on my Duet powered machine 350.

That's useful information. I'll update some profiles to reflect that.

CCS86, 2021-12-15:
Great!

Once we have a calibration standard, it will be easy to generate really good profile data.

CCS86, 2022-08-15:
Any progress on this one @supermerill?