supermerill/SuperSlicer #2073: Extra deretraction distance based on travel distance

mirokymac, 2021-12-17:
**Is your feature request related to a problem? Please describe.**
As printing the high flowable filament, the amount of filament goos out of the nozzle
is related to the travel time. Thus, on short travel, the extra deretraction distance will
be considerably small, while big EDD is require is needed.

**Describe the solution you'd like**
3 changes on GUI:
 1 add check box <Extra deretraction distance based on travel distance> 
 2 add text box <Linear extra deretraction distance factor> avaibility base on 1
 3 add text box <Ignore EDDT while travel small then xxx> avaibility base on 1

**Describe how it would work**
 if `enable EDDT` and `travel distance` > `<Ignore EDDT while travel small then xxx>`:
     `deretraction distance = EDDT factor * travel distance`

supermerill, 2021-12-17:
duplicate of #734
i guess?

mirokymac, 2021-12-17:
yes. I think this can be cloed now.

closed 2021-12-17 by mirokymac